FreeCAD/FreeCAD #15548: CAM Simulation of a cutting move is too fast

luxapana, 2024-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Version: 0.21.2
OS: Ubuntu 22.04

Problem:
When trying to simulate the CAM for a simple cut, the movement of the cutter is instantaneous during a cutting move. Please see the video attached. 

[Screencast from 07-21-2024 02:01:56 PM.webm](https://github.com/user-attachments/assets/a23ca49d-5404-4490-a0c0-03efffa4aa03)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-21:
you can test the new simulator in the dev builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
@sliptonic is the "old" simulator being removed for 1.0? I'd close this issue then.

luzpaz, 2024-07-24:
@shaise do you mind weighing in ?

shaise, 2024-07-24:
Yes, the new simulator is much better in this case. The old one is rather limited.
Hopefully V1.0 will merge all the latest features.
Regarding old simulator, it is not going to be removed yet, until we are sure the new one is working well.

luzpaz, 2024-07-24:
Closing as no changes will be made to the old simulator moving forward.

closed 2024-07-24 by luzpaz