FreeCAD/FreeCAD #15549: Sketcher: Crash when box-selecting normal and external geometry

maxwxyz, 2024-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD crashes instantly when I box-select a combination of external and normal geometry, drawn on top of each other. It only crashes when I select it directly after creation, so if I redraw the external geometry on top with normal geometry, close the sketch and save and open the file again, the box selection works as normal. It just crashes in the same active sketch as the external and normal geometry was drawn.
[image]

Demo file:
[Sketcher-crash.FCStd.zip](https://github.com/user-attachments/files/16323759/Sketcher-crash.FCStd.zip)

Steps to reproduce:
1. Open the demo file
2. Open the sketch in the second body. Select all geometry and delete everything.
3. Get the "External Geometry" or the two arcs (left and right) of the sub-shapebinder
4. Redraw a slot by referencing the external geometry (concentric centerpoints left and right and finish with a concentric point at the outer point on the tangent arc point)
5. Box select the entire geometry

Demo video:

https://github.com/user-attachments/assets/d744b5c6-a73d-43cd-9ed5-b3c6fd4e6904

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.22
  * Curves 0.6.41
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-21:
@PaddleStroke FYI
@bgbsww could this be a result of TNP-fixes?

bgbsww, 2024-07-21:
Highly likely, as this has to do with ExternalGeo being correct in all cases, and it changes as part of the TNP work.  Every time I've seen this category of crash, it is because of attempts to access invalid ExternalGeo while selecting.

Unfortunately, due to the save/load issues, any testing based on loading an example file rather than building from scratch is likely to be invalid.

There is a large series of PRs making its way through to address the save/load issues, and they fix the ExternalGeo issues, so I suspect this will go away once they are all merged.

PaddleStroke, 2024-07-22:
An issue was posted on ondsel tracker and it seems to be related : https://github.com/Ondsel-Development/FreeCAD/issues/108

ShayGus, 2024-07-27:
Can confirm this crash on Windows 11 Pro 23H2
FreeCAD 0.21.2

maxwxyz, 2024-07-29:
This should be fixed with #15575 
Please test again with the latest weekly tomorrow: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

FEA-eng, 2024-07-30:
No issue for me anymore (apart from that sketcher error and disappearing slot reported in #15494) but maybe it's worth rechecking on Mac. Windows usually has no crashes, just errors.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

maxwxyz, 2024-07-30:
I cannot delete external geometry now in today's weekly, can you confirm with this file @FEA-eng ?

bgbsww, 2024-07-30:
Confirmed, I can't either.  That will be something missed in deleting
external geo in SketchObject.cpp after the switch from instance variable to
property.

bgbsww, 2024-07-31:
Extra confirmed with a debug run, there is a code problem in SketchObject::delExternal that needs comparison and fixing.  Should be straightforward, but is around #7 on my current list if someone else wants to take this one; just compare against the LS3 code.

bgbsww, 2024-07-31:
#15712 will fix the delete external geo issue.

PR #15575 (2024-07-29): Toponaming:  Final save/restore changes
Appear to be fixed:  #15549  #15380 #15484   Each of these should get verified before being closed which is why no direct Fix link here.  I'm pretty sure on these ( my tests worked ), and very hopeful on 15549.