FreeCAD/FreeCAD #15550: PartDesign: TNP for same edge - works in 0.21

maxwxyz, 2024-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I finally made a file where I can reliably reproduce the TNP issue which I've encountered so often lately.
I have a simple sketch (slot). 
This profile is extruded via a pad in PartDesign and then a fillet is applied to one edge (Edge11). 
As soon as I change the radius of the slot in the sketch, the fillet is broken. 
There are not more edges or vertices in the scene than before, the sketch is simply larger.
The fillet command then shows `?Edge11` and when I select the exact same edge in the fillet tool again, it is called `Edge11` again. So there shouldn't be an issue IMO. To remove the bug, I need to manually delete `?Edge11` from the task dialog as I cannot deselect it in the 3D view and add the same edge `Edge11` manually again.
The same works in the stable 0.21.2 version without a problem, as the edge is the same name.

Demo file (with the already changed sketch, so the fillet is already broken):
[TNP-issue.FCStd.zip](https://github.com/user-attachments/files/16323958/TNP-issue.FCStd.zip)

Demo video:

https://github.com/user-attachments/assets/fc5b131f-9da8-4232-92ff-c89c053c4a98

[image]

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.22
  * Curves 0.6.41
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-21:
@bgbsww FYI

maxwxyz, 2024-07-21:
Here is the demo when I manually reset it to the right edge and then change the sketch again, it will break again as soon as the sketch is edited. On the second time when resizing, it actually works.

https://github.com/user-attachments/assets/d973507b-b8cb-4886-9070-2136d5d57e23

maxwxyz, 2024-07-21:
@bgbsww If I fully recompute the document before changing the sketch, it has no issues at all...

maxwxyz, 2024-07-21:
So is this a duplicate to an existing issue and should be closed or left open to check if it will work after the next fixes?

bgbsww, 2024-07-21:
It is in all probability a dup of anything that's complaining about '?'
missing element names.  I suppose maybe we should be grouping these all
together.   Note that a run of the tests will throw up a fair clump of
missing warnings, which as it turns out seem to be ultimately fatal.

bgbsww, 2024-07-25:
Probably worth mentioning that an initial full recompute of a file immediately after loading it appears to help with some issues, but 0.22 DEV is still in the use at your own risk territory and can break models.  We are close.

maxwxyz, 2024-07-29:
This should be fixed with #15575 
Please test again with the latest weekly tomorrow: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

FEA-eng, 2024-07-30:
This seems to be fixed but please recheck yourself (in a fresh file) just in case.

[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

maxwxyz, 2024-07-30:
works, thanks!

closed 2024-07-30 by maxwxyz