FreeCAD/FreeCAD #15552: VarSet: Qt warning when using the "Show variable sets" option in the Expression editor

Roy-043, 2024-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

You can now check the "Show variable sets" checkbox in the Expression editor. If you do this there is a Qt warning however.

`14:07:49  QWindowsWindow::setGeometry: Unable to set geometry 414x292+102+443 (frame: 430x331+94+412) on QWidgetWindow/"DlgExpressionInputWindow" on "\\.\DISPLAY1". Resulting geometry: 414x293+102+443 (frame: 430x332+94+412) margins: 8, 31, 8, 8 minimum size: 300x292 MINMAXINFO maxSize=0,0 maxpos=0,0 mintrack=316,331 maxtrack=0,0)`

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pieterhijma, 2024-07-23:
I can confirm this and I think it is Windows related. I don't have the warning on Linux but I'm running Qt 6, so it could also be a Qt 5 issue.

luzpaz, 2024-10-23:
Anymore traction on this issue?

luzpaz, 2024-10-31:
@johnocfii any chance you can test this on macOS ?

JohnOCFII, 2024-10-31:
> @JohnOCFII any chance you can test this on macOS ?

Assuming I'm testing appropriately (I've never used VarSets before this test) I'm not seeing a problem.  

Also, looking at the text of the message in the OP issue, is the problem that there is a message at all, or that there is a message in the Report Window that should really be a user-facing dialog regarding minimum or maximum window size?

```
OS: macOS 15.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git)
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.51
```
[image]

Roy-043, 2024-12-12:
The problem still exists for me.
```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

pieterhijma, 2025-08-31:
I can reproduce on:

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

However, I don't fully understand what triggers the error.

pieterhijma, 2025-08-31:
I can't reproduce on this version:

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/100.531/2
```

This is likely because Qt has been upgraded. @Roy-043, can you try to trigger this with a weekly that has Qt 6, like the one above?

Roy-043, 2025-09-01:
I can't test on Windows anymore. But if you can't reproduce with a new Qt version on Windows then this can be closed.

closed 2025-09-01 by Roy-043