FreeCAD/FreeCAD #15555: VarSet: confusing input handling in "Show variable sets" option in the Expression editor

Roy-043, 2024-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you check the  "Show variable sets" option in the Expression editor you can add a property to an existing VarSet. The first two inputs are confusing.

In the "Variable Set" dropdown list you can select a group, which is perhaps not completely wrong, but does result in confusing feedback. The user sees "Variable Set: GroupName" in the UI, which is misleading.

The Group input does not display a dropdown list. Which is inconsistent when compared to the "Add a property" dialog box of the Std_VarSet command.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kriswilk, 2024-10-08:
> In the "Variable Set" dropdown list you can select a group, which is perhaps not completely wrong, but does result in confusing feedback. The user sees "Variable Set: GroupName" in the UI, which is misleading.

To clarify this issue: choosing a group name from the "Variable Set" tree incorrectly populates the group name in BOTH the "Variable Set" and "Group Name" fields. For instance:

[image]

If instead you select the VarSet from the tree, it is populated correctly, but leaves the group name blank with no way to select it except by typing it manually:

[image]

pieterhijma, 2025-04-02:
I'm looking into this one and https://github.com/FreeCAD/FreeCAD/issues/17075. I agree with all the suggestions. Just to be sure, @FreeCAD/design-working-group, could you please take a look at this (and in particular the suggestions in #17075) as well?

Roy-043, 2025-07-13:
Still relevant.

closed 2025-08-27 by kadet1090