FreeCAD/FreeCAD #15558: Expression: Cannot use boolean value directly in conditional

Roy-043, 2024-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

You cannot use a Boolean value directly as the condition in a conditional expression.

You have to use this syntax:
`VarSet.MyBool == 1 ? 10 mm : 15 mm`

Where I would expect this to be valid as well:
`VarSet.MyBool ? 10 mm : 15 mm`

And this should also work IMO:
`1 ? 10 mm : 15 mm`

Documentation:
https://wiki.freecad.org/Expressions#Conditional_expressions

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-21:
@Roy-043 is this the same as https://github.com/FreeCAD/FreeCAD/issues/8008 ?

Roy-043, 2024-07-24:
I did find #8008 before creating this issue. It is obviously related but different.

jffmichi, 2025-07-21:
#22630 is introducing an `if` function as a workaround for this issue. This issue should not be closed by that PR.

mnesarco, 2025-07-21:
The ternary operator should not be removed, as we want to keep backwards compatibility and do not break existing models. It can be deprecated (supported with a warning).

Changing the ternary operator to support direct use of `cond` rule as `exp` requires some refactoring of the base grammar. Its not impossible of course but needs extensive and well designed tests because it can be tricky to ensure the semantic remains unchanged.

mnesarco, 2025-07-21:
I agree that #22630 provides an alternative conditional mechanism, not a direct solution to the ternary operator limitation itself (this issue)