FreeCAD/FreeCAD #15559: VarSet: Sugggested UX improvements

hooperbloob, 2024-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Some minor tweaks to the VarSet UI could be helpful:

As of release 38261, VarSet items shown in the model tree have little UI value. This could be improved by showing the label followed by the number of entries it might have:   <varSet name> ( <total # of entries> : < # of hidden entries>, etc).

I would suggest showing the actual values in the tree as child nodes also. Users could select to delete them or move them between different VarSet containers.  If you do show the values as tree nodes, render them to denote their configuration. I.e. italics for read-only, light grey for hidden, etc.  Maybe use icons as well.  Any unreferenced values could be shown with slashthrough text.  Whichever rendering conventions you use should be consistent with any pre-existing ones in FreeCAD and play nicely with style sheets.

The last entry in VarSet popup menu shows 'Status' for what is essentially an entry point for boolean settings.  I would rename that menu label to 'Configuration'.

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * BIM021 2021.12.0
  * fasteners 0.5.24
  * sheetmetal 0.4.12
  * Assembly4 0.50.15
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-21:
@pieterhijma FYI

luzpaz, 2025-04-20:
@pieterhijma should this ticket be split up into separate issues ?

pieterhijma, 2025-04-21:
Yes, could be useful.

pieterhijma, 2025-08-27:
@hooperbloob, on 1. show # entries in VarSet label. I think this could be useful, but I'm not aware of any document object (and a VarSet is a document object) doing this kind of label changing. It would make referencing by label challenging as well.

pieterhijma, 2025-08-27:
On 3. I personally prefer to keep it at Status.

luzpaz, 2026-01-01:
Split ticket. Closing