FreeCAD/FreeCAD #15560: VarSet: Cannot delete entries directly and menu inconsistency

hooperbloob, 2024-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While you can delete VarSet entries via the context menu, you're not able to do so directly via the delete key - at least not under MacOS.

I would also suggest modifying the context menu label from 'Delete property' to just 'Delete', and include the same garbage can icon beside it. This would make it consistent with the others.

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * BIM021 2021.12.0
  * fasteners 0.5.24
  * sheetmetal 0.4.12
  * Assembly4 0.50.15
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-21:
@pieterhijma FYI

pieterhijma, 2025-04-02:
I'm looking into this. Technically, this is not a VarSet issue, but a Property View issue, but I can resolve this as part of the VarSet UX project.

I believe that the point is that the context menu for document objects (in the tree) has icons besides each action, whereas this context menu doesn't. The issue mentions "Delete property" whereas on my system it is "Remove property". The suggestion is to simplify it to "Delete" or "Remove". However, there is also the action "Rename property group", and perhaps there will also be a "Rename property" in the near future, so because of that I find it more logical to have "Delete property" instead of simply "Delete". 

As for the icon. I would in favor of not having icons here because this specific action would stand out against the other actions that don't have an icon.

So, I propose:
- Change "Remove property" to "Delete property"
- Add a keyboard binding for the Delete key for this action.

Perhaps the @FreeCAD/design-working-group can take a look at this?

pierreporte, 2025-04-04:
I agree that the keyboard binding would be a good thing. Naming should be consistent.

closed 2025-09-01 by kadet1090