FreeCAD/FreeCAD #15561: UX: Acknowledge copies to clipboard

hooperbloob, 2024-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the FreeCAD About box you're able to copy the configuration to the clipboard but when the user clicks the button, nothing confirms that this happened. Most applications I've used will acknowledge this with a quick popup message: 'Copied to clipboard'.  We need the same thing.

There could be similar copy-to-clipboard facilities in FreeCAD that have the same deficiency.

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * BIM021 2021.12.0
  * fasteners 0.5.24
  * sheetmetal 0.4.12
  * Assembly4 0.50.15
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-30:
What about a message appearing in the Report View instead ? Popups get some users upset because they don't like it (considered spam)

hooperbloob, 2024-07-30:
It doesn't need to be a full blown popup.  Heck, even if you just change the button label from 'Copy to clipboard'  to 'Copied!'  for two seconds and then revert that'd be enough.

furgo16, 2024-07-31:
> What about a message appearing in the Report View instead ?

Or in the notification area

yashathwani, 2024-08-01:
Can I work on this issue ?

luzpaz, 2024-08-02:
Sure! Feel free to propose a fix. Note: it will probably be postponed till post 1.0 release since we're in a feature freeze, but nevertheless, have at it!

luzpaz, 2024-08-30:
@yashathwani ping

BMaxV, 2024-11-03:
@hyarion recommended I take this one as a first issue, so I will have a go at it. :)

hyarion, 2024-11-05:
> Or in the notification area

The notification area is only shown when the main window is focused. When about window is opened, the user needs to click the button at the bottom to see it.

> It doesn't need to be a full blown popup. Heck, even if you just change the button label from 'Copy to clipboard' to 'Copied!' for two seconds and then revert that'd be enough.

This could work.

closed 2024-11-25 by chennes