FreeCAD/FreeCAD #15564: Spreadsheet cell can't reference value of another cell by alias

nazar-pc, 2024-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Very simply workflow:
* create a spreadsheet
* set alias for a cell as `AliasName`
* try to use `=AliasName` in another cell

Despite FreeCAD autocompleting it and even simplifying more complex references down to `=AliasName`, it prints confusing error instead of succeeding:
```
20:29:53  <Spreadsheet> Cell.cpp(668): Configuration#Spreadsheet001.F2: Property 'AliasName' not found in 'AliasName'
20:29:53  Case: One or more cells failed contains errors.
```

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git) AppImage
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * parts_library
  * Curves 0.6.40
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-21:
It works on my side.
Create a cell alias for a cell `test`, enter in another cell `=test` and it works.

```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.22
  * Curves 0.6.41
```

nazar-pc, 2024-07-21:
Interesting, it didn't work in one of the spreadsheets, but when I created a new spreadsheet it did work fine. I closed sketch in another document (thinking it may affect things somehow), but as the result FreeCAD crashed (probably one of those segfaults again). After reopening the document it works just fine.

No idea how it ended up in that situation, but it certainly didn't work back then and it absolutely works fine now. Same version of FreeCAD, same document.

closed 2024-07-21 by maxwxyz

xtemp09, 2024-07-21:
If I create an alias for an empty cell, _FreeCAD_ shows this message. If the cell is not empty, it does not.

> 02:03:03  <Spreadsheet> Cell.cpp(668): Unnamed#Spreadsheet.C2: Property 'test1' not found in 'test1'
> 02:03:03  Spreadsheet: One or more cells failed contains errors.

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38275 (Git)
Build type: Debug
Branch: main
Hash: 41d67f8e108a77ec0b98a9ca47d85575eb43476c
Python 3.10.12, Qt 5.15.14, Coin 4.0.0, Vtk , OCC 7.5.1
Locale: English/United States (en_US)
```