FreeCAD/FreeCAD #15567: Measurement: No option for diameter?

JohnOCFII, 2024-07-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The new Measurement system is looking better and better -- surprised when I was looking for diameter, I received radius -- and didn't see an option in the dropdown for diameter.  Granted, diameter is just `2 x radius`, but I use computers for a reason -- to do the simple stuff!
[image]

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.7.3
```
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-22:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/13724

closed 2024-07-22 by maxwxyz

JohnOCFII, 2024-07-22:
> Duplicate of #13724

OK -- and I did see that entry, but created this as it was just the small matter of displaying diameter, (radius worked fine for this circle), and not the larger issue raised in the other item.  But yes, if the other were solved completely, this could be as well -- just figured this tiny change might happen first.