FreeCAD/FreeCAD #15570: Hole operation only goes one direction though configured as midplane+through all

knipknap, 2024-07-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

The operation succeeds, but only does in one direction instead of mid-plane.

Steps to reproduce:
1. Open the file: [bearing.zip](https://github.com/user-attachments/files/16332305/bearing.zip)
2. Clock on the Hole operation
3. Note that "Through all" and "midplane" are selected.
4. Note that the hole does not go in both directions.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git) AppImage
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * DynamicData 2.62.0
  * btl 0.9.9
  * WebTools
  * toSketch
  * fasteners 0.4.62
  * lattice2 1.0.0
  * freecad.gears 1.2.0
  * sheetmetal 0.4.23
  * CycloidGearBox
  * Assembly4 0.12.3
```

### Subproject(s) affected?

PartDesign

### Anything else?

No logs are shown in the console.

The problem exists in FreeCAD 19, too. See https://forum.freecad.org/viewtopic.php?t=51903

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-28:
This works as expected. There is no symmetry command in the hole dialog. You can use the pocket dialog but then need a new sketch with your diameter and there you could select symmetric.
Not sure how valid a symmetry checkbox would be in a hole dialog where you usually make threads and features like countersunk, @pierreporte maybe you can weigh in if this is a feature.

knipknap, 2024-07-28:
I had expected it to just perform the operation twice; once in each direction.

But if it's not intended to work I think it isn't a big issue, that type of use is a corner case and could be done just as well using a mirror operation.

In that case I think it would be great to remove the unused/not working property, as it would have saved me some time trying to track down the (assumed) problem.

pierreporte, 2024-07-28:
I agree with @maxwxyz. It could be valid for the most simple holes, but once they are countersunk or counterbored, a symmetry wouldn’t make sense in my opinion. On the other hand, it would make sense for threads but implementing the feature just for these corner cases may not be worth it.

knipknap, 2024-07-28:
Just for the record, I think it _can_ make sense to counterbore in both directions. An example is piece is one with flanges, to be counterbored from the inside using a single sketch:

[image]

But yes, it is a corner case and a mirror transformation can achieve the same thing with not much effort.

luzpaz, 2026-09-06:
This looks like a duplicate of https://github.com/FreeCAD/FreeCAD/issues/19764  
If it is, then @alfrix wrote a response why this may never become a feature (https://github.com/FreeCAD/FreeCAD/issues/19764#issuecomment-3053281591)

closed 2026-09-06 by maxwxyz