FreeCAD/FreeCAD #15576: Part Design - 3D View/Task dialog issue in Pad/Pocket with Up to face Type

marcuspollio, 2024-07-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the Part Design workbench, with the Pad and Pocket tools based on a selected sketch, the Type parameters has issues when switching between possible types, choosing "Up to face" once and choosing another type.

1) In the Start Dialog, choose the Parametric Part option (Part Design WB)
2) In the default Body, create a simple base sketch on an origin plane (e.g. XY plane).
3) Pad this sketch so you have a solid shape.
4) Create another simple sketch, either from an origin plane (then offset it based on the Pad length), or from a selected face of the Pad, or a datum plane. Your choice =D
5) Select this second sketch and choose the Pad or Pocket tool.
6) In the task dialog of the Pad or Pocket tool, the default Type is "Dimension".
7) The 3D View shows the resulting operation. See the first picture below.
8) Switch between the different available Types one after the other, coming back to the default "Dimension" Type to see the default behavior described in the previous point 7.
9) If switching to "Up to face" Type and back to "Dimension" Type, the behavior described above at point 7 is lost. See the second picture below.

Default behavior with Dimension Type :

[image]

Behavior after switching to Up to face Type and back to Dimension Type :

[image]

### Full version info

```shell
OS: GNU/Linux Ubuntu 22.04 LTS (DE: Gnome/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38317 (Git)
Build type: Release
Branch: main
Hash: ec586668d9794afdfe574887f16c215f744ee8c0
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.6.3
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-28:
@FlachyJoe FYI

marcuspollio, 2024-12-05:
Hi !
Just tested with a recent build, the issue is still present.
<details><summary>About info</summary>

'''
OS: GNU/Linux Ubuntu 22.04 LTS (DE:GNOME/pop/xcb)
Architecture: x86_64
Version: 1.1.0dev.39422 (Git)
Build type: Release
Branch: main
Hash: 41ca58bf7c82881740288f3059ed9bac0efd6447
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.6.3
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
'''
</details>

tarman3, 2024-12-05:
Confirm. But can reproduce it only one time.
Next try this issue not happened.

marcuspollio, 2024-12-05:
> Confirm.

Hi, thanks for checking. Yeah, it's not a big deal anyways, let's see when transparent previews lands.

luzpaz, 2025-08-09:
@kadet1090 does https://github.com/FreeCAD/FreeCAD/pull/17249 impact this ?

marcuspollio, 2025-08-11:
@luzpaz Thanks for keeping us in the loop ;D

I just tested with

```
Version: 1.1.0dev.42894 (Git)
Build date: 2025/08/10 15:35:44
Build type: Release
Branch: main
Hash: 1885cce10eada09cf523d754348b2bc7e2b66a9b
```
And cannot reproduce this issue anymore.
After the point 9. (choosing `Up to face` or `Up to shape` and going back to `Dimension`) we get the expected result:

[image]

There are some error messages when switching to `Two dimensions` afterwards, but that's another issue.

I guess we can close this issue if @kadet1090 also confirms the above. TIA!

closed 2025-08-14 by kadet1090