FreeCAD/FreeCAD #15578: [nightly/stable] Part design Pad is not working, only one shape extruded

simonbuehler, 2024-07-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi there,
while fiddling to recreate a broken keychain ghost I tried to add some support channels up to the existing shape (face) but even the simple 10mm extrusion just uses one of the multiple circles in the sketch.
sketch 
[image]

pad
[image]

original just for understanding that the aim would be to extrude up to the face
[image]

also switched to nightly but still not working, only error i see:

`09:30:59  SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices`

please let me know what to do to debug this issue, thanks!

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.60m
  * Assembly4 0.50.2
  * Curves 0.6.23
  * fasteners 0.5.24
  * Freecad-Built-in-themes-beta 1.0.0
  * FreeCAD_BearingsWB 0.0.2
  * OpticsWorkbench 1.0.9
  * ThreadProfile 1.89.0
```

### Subproject(s) affected?

PartDesign

### Anything else?

the project file 
[geistv21.zip](https://github.com/user-attachments/files/16345399/geistv21.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-28:
Is it that what you want:
[image]

If so, it works with up to face, up to last or up to shape as demonstrated.

closed 2024-07-28 by maxwxyz

simonbuehler, 2024-07-28:
It seems to only work for the first one [
[image]
](url)

maxwxyz, 2024-07-28:
maybe because of the intersection between the line of the outer shape on the last one (could be OCC related). Did you do it with up to face or up to shape or to next?
@FlachyJoe FYI

simonbuehler, 2024-07-28:
I think  the thickness operation leads to the errors as when I do a Boolean cut instead to remove the inner shape it works

simonbuehler, 2024-07-28:
> maybe because of the intersection between the line of the outer shape on the last one (could be OCC related). Did you do it with up to face or up to shape or to next? @FlachyJoe FYI

I tried all variants, just adding more shapes to the pad sketch shows the problem also at other positions