FreeCAD/FreeCAD #15579: Copying parts of an expression sketch after pasting breaks the expression (Ctrl+C / Ctrl+V).

kaiwas, 2024-07-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If we copy an element with an assigned expression and paste it, then the inserted element will display the correct constraint value, but it will no longer be an expression.
However, there is no way to identify this in the 3D window. To see this you need to open constraints.

https://github.com/user-attachments/assets/a345bb9b-8021-48a7-998a-6dfdc5ffea29
[Copy-Paste.FCStd.del_this_and_zip.zip](https://github.com/user-attachments/files/16348460/Copy-Paste.FCStd.del_this_and_zip.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git) AppImage
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.7.3
  * PitchedRoof
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-07-23:
New transformation tools also break expressions.

https://github.com/user-attachments/assets/79f7255d-5d4a-4af8-acb4-e7fac9ad9303

maxwxyz, 2024-07-24:
Is this working in the stable version?

kaiwas, 2024-07-24:
> Is this working in the stable version?

No.
It seems there has never been functionality for “saving” expressions. The copy tools in version 0.21 also "break" expressions (convert them to simple numbers).

https://github.com/user-attachments/assets/605b0dc3-d73d-433f-8d37-ab6377206c45

kaiwas, 2024-07-24:
If you try to formulate the most negative feature of the program in a short phrase, then I would say that it is “Loss of previously entered data.” This problem occurs in many places. (In my opinion). I assume that an adult CAD system should not have such a disadvantage. Data ever entered by the user must be stable and not lost unless the user wants it.

kaiwas, 2024-12-04:
The problem is still relevant

https://github.com/user-attachments/assets/7d909a00-e944-46a5-9e1f-5c5803820c79

```

OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * freecad.gears.backup1731915834.4791615 1.3.0 (Disabled)
  * addFC 0.7.5
  * freecad.gears 1.3.0
  * sheetmetal 0.5.10
  * OpenTheme 2024.9.1

```

maxwxyz, 2025-07-02:
I think it is not that easy as you can copy sketches from other documents or from the web or any other place where the expressions or variables do not exist in the document you are pasting them.

kaiwas, 2025-07-02:
> I think it is not that easy as you can copy sketches from other documents or from the web or any other place where the expressions or variables do not exist in the document you are pasting them.

Yes. I agree. There is something to think about.
But! Much more importantly - all the losses happen unnoticed. Perhaps at least a pop-up warning is required that "Expressions were converted to a number".

kaiwas, 2025-07-02:
> can copy sketches from other documents or from the web or any other place

By the way, in my opinion this is not a problem, if expressions are used somewhere - they should be saved. It would even be convenient.
But for now - silent data loss occurs.

DoctorWiz, 2026-02-24:
My two cents: This certainly seems like a bug and not the expected behavior.
The "Export" dialog when I 'copy' defaults to making a copy of the [dependent] varset as well.  But whether I leave that checked, or uncheck it, it still does not copy the variable references, either way.  And you would think if it makes a copy of the varset also, it would reference that one.  But nope.  And if pasting within the same project, you are most likely not want a new copy of the varset which then becomes just a snapshot of the first one at that point-in-time, you'd want the reference the original so you don't have to change values in multiple places.

But hey, FreeCAD is still awesome anyway-- thanks devs!

kaiwas, 2026-09-03:
still relevant

nyq, 2026-09-15:
Since #32697 got hastily closed, I am mentioning it here: not only expressions, but also the **names of the constraints** are lost during copy-paste.

ardugitter, 2026-10-07:
Hello. Shall I do a PR to try and fix this?

I just filed a feature that was exactly this bug (feature has been closed).

nyq, 2026-10-07:
> Hello. Shall I do a PR to try and fix this?
> 
> I just filed a feature that was exactly this bug (feature has been closed).

Fixing this would be very useful and greatly appreciated!

ardugitter, 2026-10-07:
I am looking at this now. I managed to get FreeCAD building and the bug is present in the very latest unstable version.

So, if I understand this correctly, the issue is really in pasting as the copied data is all present and correct. When you copy, FreeCAD actually creates a Python script that when run later, will recreate the original object except when that script is parsed, somehow the expressions are evaluated and lost instead of being pasted identically to the source object.

I am not an expert on constraint names (don't really use them), but surely you don't want them copied since they are unique to the sketch? Are you talking about pasting into a different sketch which doesn't already have those names assigned to something?

BTW, I am proposing that when pasting, any problems pasting the original expression just means the existing behaviour stays as it is, plus you get some red error text in the console as you do at present when e.g. wire not closed. I think this is better than an annoying pop up. 

Also, when pasting, how about the pasted object appears in the sketch where the mouse pointer is? It is confusing at present with it getting pasted exactly over the original object so it can't be seen.

nyq, 2026-10-08:
> I am not an expert on constraint names (don't really use them), but surely you don't want them copied since they are unique to the sketch? Are you talking about pasting into a different sketch which doesn't already have those names assigned to something?
> 
Constraint names are not global names, they are local to sketch and do not show up in the project tree. This is true for all objects inside the sketch - the inner components of the sketch are not exposed to the project tree. This is different from most of other names in FreeCAD projects, like the names of the origins and axes - they are global. So, because constraint names are local names, they should be copy-pasted to another sketch as is, without any renaming, unless, of course, there is a name conflict with existing constraint in the target sketch.

ardugitter, 2026-10-08:
> Constraint names are not global names, they are local to sketch and do not show up in the project tree. This is true for all objects inside the sketch - the inner components of the sketch are not [...]

Yes, constraint names should be copied. But if an expression in the object uses constraint names not already in the pasted object itself, the expression will be evaluated to its numerical value with a warning.