FreeCAD/FreeCAD #15582: Distance joint in assembly does not solve correctly - moves grounded part

tibfel, 2024-07-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am making a simple table project. I have a table top and a leg. 
[image].
Table top is grounded. 
[image]
The grounded part should remain its position and the table leg should mate the table top. However the joint changes the orientation of the grounded part to this:
[image]
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Slovak/Slovakia (sk_SK)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.7.3
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-23:
@tibfel did you try this with the latest dev version from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
There were some recent fixes iirc.
If the issue still persists in the latest weekly, please provide a demo file.

tibfel, 2024-07-24:
@maxwxyz Yes, I have used the latest weekly build. 

Here are the files. 
[Issue.zip](https://github.com/user-attachments/files/16358244/Issue.zip)
When you remove the distance joint, the tabletop reverts into the correct horizontal position. Then when you add the distance joint again, it aligns itself with the unconstrained leg. The correct solution would be the other way around. The unconstrained leg should align with the grounded tabletop.

maxwxyz, 2024-07-24:
@tibfel I cannot reproduce.
The grounded part dows not move, the distance joint is always correct and removing the distance joint does not change the orientation of the grounded part.

```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.22
  * Curves 0.6.41

```

tibfel, 2024-07-25:
@maxwxyz I have tried to investigate further and I have found out that the selection order matters. If you want to apply a distance joint and you first select grounded object and then unconstrained object, it solves correctly. However, when you first select unconstrained object and then grounded object, the grounded object aligns according to the current orientation of unconstrained object. 

See video:
1. Correct selection order : grounded object, then unconstrained object
[image]

3. Incorrect selection order: unconstrained object, then grounded object
[image]

maxwxyz, 2024-07-25:
@PaddleStroke FYI

PaddleStroke, 2024-07-25:
Thanks for the report.
There's a big change around the corner in assembly's internal. (https://github.com/FreeCAD/FreeCAD/pull/15384)
So this bug may be gone since this PR is changing just so much.

I will test your file with this new branch and see if the bug is still there. And will address it.

luzpaz, 2024-08-17:
@PaddleStroke did you get a chance test on new branch ?

tibfel, 2024-08-17:
I have tested it on the newest build at it does the same strange behaviour as before. You can download Issue.zip file from the first post and try it yourself, too. When I select the grounded part first, the joint solves correctly, but when I select the free part first, then both of them rotate in a weird way. 

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git)
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Slovak/Slovakia (sk_SK)
Installed mods: 
  * OpenTheme 2024.8.15
  * Render 2024.8.4
  * sheetmetal 0.4.24

tibfel, 2024-08-17:
I have also found out that those original parts become black when I insert them into the assembly (maybe some internal colour coding for bad/old files). When I insert other parts created in a newer dev version, the same behaviour persists :(

tibfel, 2024-08-24:
I can confirm that the issue is fixed in today's version (38506).

closed 2024-08-24 by maxwxyz