FreeCAD/FreeCAD #15583: OSError: [Errno 22] Invalid argument in AddonManager

Foadsf, 2024-07-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm encountering an issue with the AddonManager in FreeCAD 0.20. When it tries to fill in details from the wiki, I receive the following error:

```
OSError: [Errno 22] Invalid argument: 'C:\\Users\\foobar\\AppData\\Local\\Temp\\FreeCAD\\Cache\\AddonManager\\MacroIcons\\120px-FCBmpImportScreenshot.png?20210923163201'
```

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * GDT
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

None

### Anything else?

#### Possible Solution
Sanitize the file names by removing or replacing illegal characters before attempting to open them. For example, a function to remove illegal characters could be used:

```python
import re

def sanitize_filename(filename):
    # Remove illegal characters from the filename
    return re.sub(r'[<>:"/\\|?*]', '', filename)

# Example usage
constructed_name = 'C:\\Users\\foobar\\AppData\\Local\\Temp\\FreeCAD\\Cache\\AddonManager\\MacroIcons\\120px-FCBmpImportScreenshot.png?20210923163201'
cleaned_name = sanitize_filename(constructed_name)

with open(cleaned_name, "wb") as f:
    # Perform file operations
    pass
```

#### Additional Information
- **FreeCAD Version**: 0.20
- **Operating System**: Windows 10

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-24:
@Foadsf latest stable is 0.21.2    
Latest dev is 0.22.xxxxx   
Can you reproduce on either one ?

Foadsf, 2024-07-24:
@luzpaz on further investigation I realized that Chocolatey had installed the newer version without uninstalling the old one. Now I have

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
```
and the problem is gone.

luzpaz, 2024-07-24:
Thanks for following up. Just out of curiosity, Is there a way to improve things on the Chocolatey side so we can avoid these issues in the future?

closed 2024-07-24 by luzpaz