FreeCAD/FreeCAD #15584: Core: Missing Interpreter Lock

pieterhijma, 2024-07-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

See https://github.com/FreeCAD/FreeCAD/issues/15431

To reproduce, use a Python debug build, create a body, a sketch, draw something and close the sketch.

<details>
<summary>stacktrace 1</summary>
<pre>
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff06a8eb3 in __pthread_kill_internal (threadid=<optimized out>, signo=6) at pthread_kill.c:78
#2  0x00007ffff0650a30 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff06384c3 in __GI_abort () at abort.c:79
#4  0x00007ffff30a9b6f in fatal_error_exit (status=status@entry=-1) at Python/pylifecycle.c:2737
#5  0x00007ffff30cc28a in fatal_error (fd=2, header=header@entry=1, prefix=prefix@entry=0x7ffff33fe670 <__func__.43.lto_priv.0> "_PyInterpreterState_GET", msg=msg@entry=0x7ffff3411fb8 "the function must be called with the GIL held, after Python initialization and before Python finalization, but the GIL is released (the current Python thread state is NULL)", status=status@entry=-1) at Python/pylifecycle.c:2918
#6  0x00007ffff30cc302 in _Py_FatalErrorFunc (func=func@entry=0x7ffff33fe670 <__func__.43.lto_priv.0> "_PyInterpreterState_GET", msg=msg@entry=0x7ffff3411fb8 "the function must be called with the GIL held, after Python initialization and before Python finalization, but the GIL is released (the current Python thread state is NULL)") at Python/pylifecycle.c:2934
#7  0x00007ffff30cd240 in _Py_EnsureFuncTstateNotNULL (func=0x7ffff33fe670 <__func__.43.lto_priv.0> "_PyInterpreterState_GET", tstate=<optimized out>) at ./Include/internal/pycore_pystate.h:108
#8  _PyInterpreterState_GET () at ./Include/internal/pycore_pystate.h:131
#9  0x00007ffff30cd2a8 in _Py_INCREF_IncRefTotal () at Objects/object.c:217
#10 0x00007ffff3e6453b in Py_INCREF (op=0x7fff8c0d71c0) at /usr/include/python3.12d/object.h:656
#11 0x00007ffff3e64490 in Py_XINCREF (op=0x7fff8c0d71c0) at /usr/include/python3.12d/object.h:788
#12 0x00007ffff3e64465 in Py::_XINCREF (op=0x7fff8c0d71c0) at /home/pieter/current/ondsel/freecad-main/src/CXX/IndirectPythonInterface.cxx:554
#13 0x00007ffff667b82b in Py::Object::Object (this=0x7fffffff8340, ob=...) at /home/pieter/current/ondsel/freecad-main/src/CXX/Python3/Objects.hxx:188
#14 0x00007ffff4f7df94 in App::PropertyPythonObject::getValue (this=0x55555a677540) at /home/pieter/current/ondsel/freecad-main/src/App/PropertyPythonObject.cpp:64
#15 0x00007fff8219ce64 in SketcherGui::ViewProviderSketch::setEditViewer (this=0x55555a676520, viewer=0x5555599737f0, ModNum=0) at /home/pieter/current/ondsel/freecad-main/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp:3389
#16 0x00007ffff6f50f89 in Gui::View3DInventorViewer::setEditingViewProvider (this=0x5555599737f0, vp=0x55555a676520, ModNum=0) at /home/pieter/current/ondsel/freecad-main/src/Gui/View3DInventorViewer.cpp:1065
#17 0x00007ffff6777c6a in Gui::Document::setEdit (this=0x55555992c600, p=0x55555a676520, ModNum=0, subname=0x7fffffff9978 "Sketch.") at /home/pieter/current/ondsel/freecad-main/src/Gui/Document.cpp:440
#18 0x00007fff82197f45 in SketcherGui::ViewProviderSketch::doubleClicked (this=0x55555a676520) at /home/pieter/current/ondsel/freecad-main/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp:2635
#19 0x00007ffff6c57c48 in Gui::TreeWidget::mouseDoubleClickEvent (this=0x555555d01ad0, event=0x7fffffffae00) at /home/pieter/current/ondsel/freecad-main/src/Gui/Tree.cpp:1781
#20 0x00007ffff1d4be93 in QWidget::event (this=0x555555d01ad0, event=0x7fffffffae00) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidget.cpp:9029
#21 0x00007ffff1de2838 in QFrame::event (this=0x555555d01ad0, e=0x7fffffffae00) at /usr/src/debug/qt6-base/qtbase/src/widgets/widgets/qframe.cpp:521
#22 0x00007ffff0d44178 in QCoreApplicationPrivate::sendThroughObjectEventFilters (receiver=receiver@entry=0x555555db6820, event=event@entry=0x7fffffffae00) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1281
#23 0x00007ffff1cfc50c in QApplicationPrivate::notify_helper (this=this@entry=0x555555a39580, receiver=receiver@entry=0x555555db6820, e=e@entry=0x7fffffffae00) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3281
#24 0x00007ffff1d019c1 in QApplication::notify (this=<optimized out>, receiver=<optimized out>, e=0x7fffffffae00) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:2765
#25 0x00007ffff685433e in Gui::GUIApplication::notify (this=0x7fffffffc2d8, receiver=0x555555db6820, event=0x7fffffffae00) at /home/pieter/current/ondsel/freecad-main/src/Gui/GuiApplication.cpp:83
#26 0x00007ffff0d44d28 in QCoreApplication::notifyInternal2 (receiver=0x555555db6820, event=0x7fffffffae00) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1142
#27 0x00007ffff0d44d7d in QCoreApplication::sendSpontaneousEvent (receiver=<optimized out>, event=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1597
#28 0x00007ffff1cf3e2d in QApplicationPrivate::sendMouseEvent (receiver=0x555555db6820, event=0x7fffffffae00, alienWidget=<optimized out>, nativeWidget=0x7fffffffc0a0, buttonDown=<optimized out>, lastMouseReceiver=..., spontaneous=true, onlyDispatchEnterLeave=false) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:2347
#29 0x00007ffff1d61fb2 in QWidgetWindow::handleMouseEvent (this=0x555557a0d790, event=event@entry=0x7fffffffb600) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidgetwindow.cpp:649
#30 0x00007ffff1d635f0 in QWidgetWindow::event (this=0x555557a0d790, event=0x7fffffffb600) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidgetwindow.cpp:263
#31 0x00007ffff1cfc51c in QApplicationPrivate::notify_helper (this=<optimized out>, receiver=0x555557a0d790, e=0x7fffffffb600) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3287
#32 0x00007ffff685433e in Gui::GUIApplication::notify (this=0x7fffffffc2d8, receiver=0x555557a0d790, event=0x7fffffffb600) at /home/pieter/current/ondsel/freecad-main/src/Gui/GuiApplication.cpp:83
#33 0x00007ffff0d44d28 in QCoreApplication::notifyInternal2 (receiver=0x555557a0d790, event=0x7fffffffb600) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1142
#34 0x00007ffff0d44d7d in QCoreApplication::sendSpontaneousEvent (receiver=<optimized out>, event=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1597
#35 0x00007ffff1376d94 in QGuiApplicationPrivate::processMouseEvent (e=0x55555976e2c0) at /usr/src/debug/qt6-base/qtbase/src/gui/kernel/qguiapplication.cpp:2367
#36 0x00007ffff13e5d64 in QWindowSystemInterface::sendWindowSystemEvents (flags=flags@entry=...) at /usr/src/debug/qt6-base/qtbase/src/gui/kernel/qwindowsysteminterface.cpp:1114
#37 0x00007fffec3f7637 in xcbSourceDispatch (source=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/plugins/platforms/xcb/qxcbeventdispatcher.cpp:57
#38 0x00007fffedf77ab9 in g_main_dispatch (context=0x7fffe4000f00) at ../glib/glib/gmain.c:3344
#39 0x00007fffedfd99e7 in g_main_context_dispatch_unlocked (context=0x7fffe4000f00) at ../glib/glib/gmain.c:4152
#40 g_main_context_iterate_unlocked.isra.0 (context=context@entry=0x7fffe4000f00, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4217
#41 0x00007fffedf76fc5 in g_main_context_iteration (context=0x7fffe4000f00, may_block=1) at ../glib/glib/gmain.c:4282
#42 0x00007ffff0fa2cbd in QEventDispatcherGlib::processEvents (this=0x5555558365e0, flags=...) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qeventdispatcher_glib.cpp:394
#43 0x00007ffff0d4f01e in QEventLoop::processEvents (this=0x7fffffffba20, flags=...) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qeventloop.cpp:100
#44 QEventLoop::exec (this=0x7fffffffba20, flags=...) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qeventloop.cpp:182
#45 0x00007ffff0d4934d in QCoreApplication::exec () at /usr/src/debug/qt6-base/qtbase/src/corelib/global/qflags.h:74
#46 0x00007ffff6673fd6 in (anonymous namespace)::tryRunEventLoop (mainApp=...) at /home/pieter/current/ondsel/freecad-main/src/Gui/Application.cpp:2063
#47 0x00007ffff66711d2 in (anonymous namespace)::runEventLoop (mainApp=...) at /home/pieter/current/ondsel/freecad-main/src/Gui/Application.cpp:2083
[... cut ...]
@wwmayer, what is the right granularity here? The common frame is 15 (for both stacktraces), but adding a lock there may lead to deadlocks perhaps. Would it be better to add locks at the `src/CXX` level?

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38317 (Git)
Build type: DEBUG
Branch: main
Hash: ec586668d9794afdfe574887f16c215f744ee8c0
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk , OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-07-23:
> what is the right granularity here?

The right place to add PyGILStateLocker is ViewProviderSketch::setEditViewer.

> The common frame is 15 (for both stacktraces), but adding a lock there may lead to deadlocks perhaps. 

No, it won't cause a deadlock. It's even possible to have nested calls of PyGILStateLocker without causing a deadlock. A deadlock can only occur if two different threads try to hold the GIL.

> Would it be better to add locks at the src/CXX level?

No. Any changes to PyCXX that cause a different behaviour compared to upstream are not allowed. This is because several Linux distributions have the policy to use an available package of a library. Now if we added such changes to our local version it would lead to hard to find bugs.

pieterhijma, 2024-08-05:
> The right place to add PyGILStateLocker is ViewProviderSketch::setEditViewer.
>
>
> No, it won't cause a deadlock. It's even possible to have nested calls of PyGILStateLocker without causing a deadlock. [...]

Right, got it.

> No. Any changes to PyCXX that cause a different behaviour compared to upstream are not allowed. This is because several Linux distributions have the policy to use an available package of a library. [...]

Right, I didn't realize it was external code (I guess my editor just jumped into the relevant function).

closed 2024-08-12 by chennes