FreeCAD/FreeCAD #15589: [feature] Ignore self intersection on subtractive operations

Anthony-Gaudino, 2024-07-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While executing some subtractive operations, for example, Subtractive Pipe, if there's an intersection during this operation, instead of subtracting it will add a solid.

It would be great if there would be a way to ignore self intersections and only subtract material regardless.

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Launcher
  * Nodes 0.1.36
  * InventorLoader 1.4.0
  * Assembly4.backup1719958896.85335 0.50.12 (Disabled)
  * Render.backup1719959316.120863 2024.6.5 (Disabled)
  * woodworking 0.21.2.33771
  * ThreadProfile.backup1719959316.6397839 1.90.0 (Disabled)
  * MeshRemodel.backup1719959315.871541 1.9.23 (Disabled)
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * Pyramids-and-Polyhedrons
  * Curves.backup1719958901.528519 0.6.36 (Disabled)
  * fasteners.backup1719958901.808955 0.5.20 (Disabled)
  * nurbs
  * sheetmetal.backup1719959316.3711638 0.4.21 (Disabled)
  * A2plus.backup1719958896.309429 0.4.64a (Disabled)
  * Manipulator 1.5.7
  * lattice2 1.0.0
  * CurvedShapes 1.0.9
  * toSketch 1.0.1
  * Curves 0.6.42
  * Assembly4 0.50.15
  * Assembly3 0.12.2
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-23:
Please add a demo file, ideally screenshots to better understand the issue. Maybe this is the same as in https://github.com/FreeCAD/FreeCAD/issues/15452

Anthony-Gaudino, 2024-07-23:
@maxwxyz I have added screenshots and a minimal file as requested.

luzpaz, 2024-07-24:
@JohnOCFII is this reproducible on latest macos ?

Anthony-Gaudino, 2024-07-24:
@luzpaz I have tested on 0.22.0dev.38079 (Git) and it still happens there.
I updated the Full version info.

luzpaz, 2024-07-24:
@Anthony-Gaudino thanks. Did you overwrite the stable version you tested this on? If so, please undo that. It helps to show all versions you've reproduced this on.

JohnOCFII, 2024-07-24:
> @JohnOCFII is this reproducible on latest macos ?

I see similar results.  I've not used subtractive pipe (except in a training session a couple of years ago) so I'm not an expert in its usage.

The report window has some interesting data: 
```
09:56:37  <App> PropertyTopoShape.cpp(431): Pending recompute for generating element map: example_subtractive_pipe#Body
09:56:37  <App> PropertyTopoShape.cpp(431): Pending recompute for generating element map: example_subtractive_pipe#Box
09:56:37  <App> PropertyTopoShape.cpp(431): Pending recompute for generating element map: example_subtractive_pipe#Box
09:56:37  <App> PropertyTopoShape.cpp(431): Pending recompute for generating element map: example_subtractive_pipe#Sketch
09:56:37  <App> PropertyTopoShape.cpp(431): Pending recompute for generating element map: example_subtractive_pipe#Sketch001
09:56:37  <App> PropertyTopoShape.cpp(431): Pending recompute for generating element map: example_subtractive_pipe#SubtractivePipe
09:56:37  <App> PropertyTopoShape.cpp(431): Pending recompute for generating element map: example_subtractive_pipe#SubtractivePipe
09:58:39  SubtractivePipe001: No spine linked
09:58:39  SubtractivePipe001: No spine linked
```

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.7.3
```

[image]

maxwxyz, 2026-04-13:
@pieterhijma is this an OCC limitation or FreeCAD code?

codeotaku, 2026-08-21:
Updating with some additional progress on self-intersections in additive/subtractive sweep operations:

The previous method was targeting global self-intersections (not local), which was only part of a potentially more robust sweep solution in FreeCAD.  Currently working on implementing local and global self-intersection handling following [this research paper](https://link.springer.com/article/10.1631/jzus.A071357), but using OCCT methods. 

A few issues/test cases that are seemingly resolved with the current implementation of the method: 

[Issue 29870](https://github.com/FreeCAD/FreeCAD/issues/29870)

[image]

[image]

[Issue 21699](https://github.com/FreeCAD/FreeCAD/issues/21699) 

[image]

[image]

[image]

[issue 15589](https://github.com/FreeCAD/FreeCAD/issues/15589) above:

[image]

codeotaku, 2026-08-30:
Porting the prototype implementation to OCC.  It's handling pretty complex self-intersection cases now (in addition to the ones I was testing above). 

[image]