FreeCAD/FreeCAD #15591: Preferences window cuts off section header text with latest weekly release

DoubleStrike, 2024-07-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The preferences dialog has section headings on the right that cutoff on Windows 10 (using a standard, not a high DPI display).  This is a with a clean from-scratch install.

Look at all the section headings under the word "Navigation" on the right pane.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DoubleStrike, 2024-07-23:
Note that this does not occur with the "Classic" theme, but that theme generally looks pretty ancient, and the newer, nicer looking themes both have this problem.

maxwxyz, 2024-07-25:
It is not just in the preferences but all dialog windows. Can reproduce on Windows 10 and 11 with HiDPI and normal DPI monitor.
Here e.g. the STEP import dialog:
[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7

```

MisterMakerNL, 2024-07-25:
This is some QT bug. Have not seen it in 6 so hooe it's fixed there.

luzpaz, 2024-07-26:
@maxwxyz can you test on a Qt6 build ?

jag45, 2024-07-26:
Hi
Try the following in the relevent Dark/Light.css file. It works for me in Windows 10.

[image]

Replace the ex with em as shown here.

[image]

Hopefully ths is ok.

MisterMakerNL, 2024-07-27:
So I don't have this issue and adding 3em results to this:
[image]

MisterMakerNL, 2024-07-27:
@jag45 Could you test if 15px works instead of using 3ex?

jag45, 2024-07-27:
@MisterMakerNL  I actually set it to 3em which I agree gives too much space.

15px appears to work OK as in this screen shot.
[image]

Personally I think it looks better with 20px - gives a bit more seperation from the box above the header

[image]

Also something like 6ex seems to work nicely. 
Am I right in thinking that as 'ex' is related to the height of the font this would automatically adjust for different font sizes rather than having a fixed pixel size?

MisterMakerNL, 2024-07-27:
3ex as was should indeed be 3 times the font, but the setting is in groupbox and the text is in the title. So this is probably an issue. Hopefulle setting it to somethine fixed like 18px would solve this issue.

jag45, 2024-07-27:
OK here is a screenshot with an 18px setting. Looks fine.

[image]

Jon

closed 2024-08-17 by maxwxyz