FreeCAD/FreeCAD #15592: Topo naming issue or a broken project file?

JDMils, 2024-07-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

My main post is in the FC forums here: https://forum.freecad.org/viewtopic.php?t=89350

Basically, if I edit a sketch dimension, a Fillet breaks most probably due to the Topo issue, however if I open/close/open the same sketch, edit the item's height and close the sketch, everything works OK.
I just need to know if this is my model that is broken so that I can rebuild it in a new project file.

I'm worried about the warning messages I'm getting similar to this one:
`10:17:51  <PropertyLinks> PropertyLinks.cpp(435): Broken__Collar_with_extension_block_and_hole_markster#Fillet.Base auto change element reference Broken__Collar_with_extension_block_and_hole_markster#Pad ;g3497;SKT;:H49e,E;:G;XTR;:H49e:7,F;:U;XTR;:H49e:7,E;:H-4a5,E.Edge45 -> ;g4597;SKT;:H49e,E;:G;XTR;:H49e:7,F;:U;XTR;:H49e:7,E;:H-4a5,E.Edge45`

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git)
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * Gridfinity 0.5.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-25:
@bgbsww this `?Edge...` seems the same as the other issues, e.g. https://github.com/FreeCAD/FreeCAD/issues/15550
If you confirm, I'll close this issue.

bgbsww, 2024-07-25:
Correct, reviewing that forum thread, this is very much the save/restore issue that should  be fixed when #15575 hits.  It is now 2 PRs away, we're almost there.

closed 2024-07-25 by maxwxyz

JDMils, 2024-07-26:
So you guys already know about the issue of Open Sketch, Close Sketch, Open Sketch, Edit Sketch, Close Sketch in order to avoid the Topo issue? I was not sure if my project was corrupt and causing this issue?