FreeCAD/FreeCAD #15593: Feecad crashes if the model is too large when writing inp file

LHY19990707, 2024-07-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I want to export inp files in freecad, and the current practice is to let the program export automatically in the FEM module. However, unfortunately, when the model is relatively large and the grid is large, whether it is exporting inp files or performing finite element analysis, the program will crash directly!

I would like to know if there is another way to export the inp file, any suggestions would be appreciated!
[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Chinese/China (zh_CN)
[/code]
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-24:
Could you provide a demo file and also try this in the latest dev version?
@FEA-eng do you know such a behavior?

FEA-eng, 2024-07-24:
> @FEA-eng do you know such a behavior?

Maybe not crashes but definitely freezes: #12706

One of the biggest problems with FEM is that there’s no way to stop meshing or analysis submission and avoid this freezing common for large models: #5914

I would definitely advise starting from a coarser mesh though.

LHY19990707, 2024-07-24:
> > @FEA-eng do you know such a behavior?
> 
> Maybe not crashes but definitely freezes: #12706也许不是崩溃，但肯定会冻结： #12706
> 
> One of the biggest problems with FEM is that there’s no way to stop meshing or analysis submission and avoid this freezing common for large models: #5914有限元模型的最大问题之一是无法停止网格划分或分析提交，也无法避免大型模型常见的这种冻结情况： #5914
> 
> I would definitely advise starting from a coarser mesh though.不过，我绝对建议从较粗的网格开始。

Thanks for your reply, FreeCAD did freeze when I was exporting inp files in a larger model, after I waited about 1-2 hours, it could output inp, but it took a long time

maxwxyz, 2024-07-24:
I'll close it than as duplicate.