FreeCAD/FreeCAD #15595: FEM error ccx binary

goodwin2208, 2024-07-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With a fresh install of v 0.21.2 on windows any launch of SolverCcx Tools gives the error: FEM: wrong ccx binary. When a calculation is submitted it fails with the error: CalculiX can not be started. Missing or incorrect CalculiX binary:  PATH/FreeCAD 0.21/bin/ccx.exe

### Full version info

```shell
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

FEM

### Anything else?

[FEM_thermal_error.zip](https://github.com/user-attachments/files/16358965/FEM_thermal_error.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-24:
could you try if you also get an issue with the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
@FEA-eng FYI

goodwin2208, 2024-07-24:
The same error is produced with dev version,

goodwin2208, 2024-07-24:
Error message:
1.1: CalculiX can not be started. Missing or incorrect CalculiX binary: PATH\FreeCAD_weekly-builds-38314-conda-Windows-x86_64-py311\bin/ccx.exe

Build info:
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)

File:
[FEM_thermal_error_dev.zip](https://github.com/user-attachments/files/16362091/FEM_thermal_error_dev.zip)

Syres916, 2024-07-24:
If this was purely a FreeCAD installation issue we would have been flooded on the forum with complaints. As that hasn't been the case, the thing that we should be referencing is the Windows _Path_ environment variable, not the Python Path to be clear.

If there's a different version of `ccx.exe` in another folder which is higher up in the _Path_ then it will take precedence over the one packaged in the Installer.
If @goodwin2208 isn't aware of how to do that then see https://www.java.com/en/download/help/path.html

You can then search through all files on the computer for `ccx.exe` and if one can be cross referenced with a folder in the _Path_ that has nothing to do with FreeCAD then that's the root cause. Users can't expect the Installer to start changing existing non-FreeCAD files as that would be against all sorts of ethics/norms.

Now purely for info but possibly something to consider, this type of scenario is one of the reasons the CfdOF workbench has a _Dependency Checker_ so when support requests are logged in the forum the workbench author can verify where all the necessary mesh, graphing etc facilities are found by the workbench before even running any of the commands. You will notice I've not specified where most of the core facilities are located and yet the output of the Dependency Checker has found them in the Path (Linux in this case).

[image]

goodwin2208, 2024-07-25:
@Syres916 the PATH I inserted as a placeholder in my post since the real path has personal information in it.  Here is the path  again with some folders renamed:
C:\Users\goodwin\Downloads\FreeCAD_weekly-builds-38314-conda-Windows-x86_64-py311\bin/ccx.exe

From the cmd line, it works! Attached is the output from it being run in cmd. 
[test.txt](https://github.com/user-attachments/files/16371696/test.txt)

I also double checked the  the preferences and tried both:
[image]
and:
[image]

Both give the same error.

luzpaz, 2024-08-14:
bumping for visibility

FEA-eng, 2024-08-24:
I still can't reproduce it. Can you try placing the ccx binary in another location where system access issues shouldn't occur (C is often problematic because of that, try e.g. another drive or Program Files if you can).

luzpaz, 2024-08-24:
@goodwin2208 :point_up_2:

hyx0329, 2024-09-07:
I can't fully reproduce the issue here, but I definitely discovered something fishy: the __invalid__ executable path will __always__ be verified upon opening the preferences menu(with a popup warning), if the path is not empty. However it doesn't affect the use of FEM workbench if `Search in known binary directories` is set.

There are 2 factors leading to the problem I discovered.

Here I use the latest weekly-build `FreeCAD_weekly-builds-38643-conda-Windows-x86_64-py311`.

## The path will always be verified in the preferences menu

Given an invalid ccx binary path in `user.cfg`, and FreeCAD is set to search known directories:

```xml
<FCParamGroup Name="Fem">
  <FCParamGroup Name="Ccx">
    <FCBool Name="UseStandardCcxLocation" Value="1"/>
    <FCText Name="ccxBinaryPath">C:\Users\invalid_path\bin/ccx.exe</FCText>
  </FCParamGroup>
</FCParamGroup>
```

You can load the FEM workbench correctly with the drop down menu, __BUT__, if you then navigate to the preferences menu, FreeCAD will still complain with a popup telling that the path is invalid, though it's not necessary.

Maybe it's better to only verify the changed fields.

## There could be an invalid executable path in the config even when the user never sets it

When it's empty in `%APPDATA%/FreeCAD`, and the user has just installed new FreeCAD and run an FEM example, FreeCAD will create a new `user.cfg` with current ccx binary path(usually the bundled one) as the value for key `ccxBinaryPath`. However it should be empty as FreeCAD [defaults to searching binaries automatically](https://github.com/FreeCAD/FreeCAD/blob/7ed1e9380acccd686a0e3573f883ac793aec3299/src/Mod/Fem/App/FemTools.cpp#L301), and setting this value makes the config incompatible with upgrades.

## Conclusion

Due to the above mechanisms, the users may be confused to find that the ccx binary path becomes invalid after a FreeCAD upgrade, while nothing has been set explicitly.

mac-the-bike, 2024-12-07:
Here is the path again
C:\Users\cgoodwin\Downloads\FreeCAD_weekly-builds-38314-conda-Windows-x86_64-py311\bin/ccx.exe

This is not a path - it is a file.

dzid26, 2025-01-24:
Check with ` ldd .\ccx.exe` if it finds relevant lib*.dll in FreeCAD directory

luzpaz, 2025-01-26:
What's the verdict here ?

FEA-eng, 2025-01-26:
> What's the verdict here ?

We are waiting for feedback from the OP.

luzpaz, 2025-01-26:
cc @goodwin2208