FreeCAD/FreeCAD #15596: TechDraw: Incorrect area measurement

Pendelf, 2024-07-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The area tool measures the area incorrectly. The measurement does not take the cutout into account.
And the footnote should not point to the cutout in the center, but to the object being measured
[image]
[Area_problem.zip](https://github.com/user-attachments/files/16359124/Area_problem.zip)

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.24
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * OpenTheme 2024.7.3
  * PieMenu 1.7.5
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-24:
Is the area currectly calculated with the stable 0.21.2 version?
Maybe related to https://github.com/FreeCAD/FreeCAD/issues/14038 @FEA-eng FYI

Syres916, 2024-07-24:
I've just read through the code on my phone, I can understand why the arrowhead is in the wrong location, looks like an oversight. With regard to the scenario of not taking into account voids this is subject to OCC version and I'm not sure which versions the PR author @pavltom used, 7.7.2 is a version Intend to avoid personally.

WandererFan, 2024-07-24:
Due to limitations in the face finding algo, faces in TechDraw have no voids.  The outer square is 1 face and the cut-out is another face.  

To get the net area, the algo will have to be modified to find all faces contained within the bounds of the selected face and subtract their areas.

Syres916, 2024-07-24:
Thanks for clarification @WandererFan I must have been dreaming as I thought in the distant past (0.20.x days) that a sketch with voids was correctly calculated but I was wrong, apologies.

WandererFan, 2024-10-10:
I don't make drawings with area dimensions, so I need some help here.

Are there situations where the gross area of a face is interesting?  

If so, then we need to have 2 "options" on the area dimension - "outer" or "perforated" (don't know the terminology).

In the current code:

If then area dimension has a 3d reference, then the value is the perforated one.
If the area dimension has a 2d reference, the the value is the outer one.

An easy "fix" is to have the 2d case return the perforated value, but this loses the ability to get the outer value.

FEA-eng, 2024-10-10:
IMO in most cases only the true area of a face (what would be colored/hatched, taking any holes into account) is needed but keeping the old behavior as optional just in case might be a good idea. Maybe it could be triggered by selecting bounding edges (like for the Part Make face tool) instead of a face itself.

The options/modes could be named e.g. "actual area" (new) and "filled area" (old).

WandererFan, 2024-10-17:
Proposed solution for the wrong value and different behaviour for 2d/3d references:
add 1 property to Dimension: "Use Actual Area", default true
change code to report same (actual center and area) values for 2d and 3d references

Issues
Old documents with Area measurements of 2d faces that have holes will now report "actual" instead of "filled".   (The old value is restored by toggling "Use Actual Area" to false).  I don't know if this is a big deal or not. 
Does not fix the "end of leader not on actual face surface" problem.   The end is located on the center of mass of the perforated face.

[image]
[image]

WandererFan, 2024-10-20:
A couple of issues with making 2d area dimensions to be aware of:

[image]