FreeCAD/FreeCAD #15599: Failure of Part WB Common between cone and pad

krushia, 2024-07-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

You'll need this
[cone-common-bug.FCStd.delete.this.extension.zip](https://github.com/user-attachments/files/16365325/cone-common-bug.FCStd.delete.this.extension.zip)

In the screenshot below, I have the cone (50% transparent) and the pad from sketch shown. The common does not produce a shape, and shows the message:

`<Part> ViewProviderExt.cpp(1299): Cannot compute Inventor representation for the shape of cone_common_bug#Common001: Bnd_Box is void`

[image]

[image]

Changing any dimension of the cone by 0.001 mm, or changing the cone angle to anything other than 360 fixes the common. I tried to reproduce this with a cone and a wedge but was unsuccessful.

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git) AppImage
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.42
  * A2plus 0.4.65
  * DynamicData 2.62.0
  * Assembly4 0.50.15
  * Defeaturing 1.2.2
  * dodo 1.0.1
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * Design456 0.0.1
  * fasteners 0.5.24
  * SearchBar 1.0.2
  * DesignSPHysics 0.7.0(15-09-2023)
  * cadquery_module
  * ose-piping
  * Nodes 0.1.36
  * 3D_Printing_Tools
  * Pyramids-and-Polyhedrons
  * workfeature (Disabled)
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.7.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.7.5
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-07-24:
Works in 0.21.2 after a recompute.

krushia, 2024-07-24:
Earliest 0.22 AppImage I have (below) is affected.

OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36904 (Git) AppImage
Build type: Release
Branch: main
Hash: 0622cb784e199053656b50854e92eff806130408
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)

Syres916, 2024-07-24:
I've narrowed it down to just over 400 commits between mid March and mid April this year:

Good:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git) AppImage 14th March 2024
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
  * CfdOF 1.27.4
```

Bad:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git) AppImage 10th April 2024
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
  * CfdOF 1.27.4

```

I'll be struggling to do a git  bisect before the weekend, it just takes up too much resources on my 12 year old machine but at least it gives someone a more narrow scope to work through.

krushia, 2024-07-24:
Something tells me it has to do with OCC 7.6.3 -> 7.7.2

krushia, 2024-07-25:
I just built current source against OCCT 7.6.3 and it works!

Syres916, 2024-07-25:
Unfortunately this still errors with OCC 7.8.1:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38331 (Git)
Build type: Release
Branch: main
Hash: e4e17cf398294c75dc5a09dac3ee6b2bb778fa52
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * A2plus 0.4.65
  * FreeCAD-themes 2024.7.24
  * CfdOF 1.27.4

```

Syres916, 2024-07-25:
I've checked some files and created new ones applying _Common_ and as long as the user avoids co-planar faces it's as stable with OCC 7.8.1 as it is with 7.6.3.

Not sure whether OCC 7.6.3 was deemed a little lax by OpenCascade with co-planar checks and it's been subsequently tightened up.

Maybe a second opinion is needed but at the very least the https://wiki.freecad.org/Part_Common could be enhanced with some advice on avoiding errors by making sure the two shapes have no faces sharing the exact space, for example a red cross on:

[image]

maxwxyz, 2024-07-29:
@adrianinsaval which OCC version is used for the 1.0 release?

FlachyJoe, 2024-07-29:
@maxwxyz As I know OCC **minimal** version is still 7.3.3 because of Ubuntu 20.04 compatibility. But no maximal version should be required.

PR #15931 (2024-08-26): Horrible hack to work around bug in OCCT7.7.2
Fix #15599.

The failure is actually in cone, and represents a bug in 7.7.2 but not 7.6.3 and not 7.8.x.  Angle values of 360.0 fail completely, 359.9999 and 359.999 partially, and 359.99 results in an ugly fix.

If it isn't clear, I hate this code, but it seems like the pragmatic best choice; the alternative is FreeCADs with 7.7.2 don't work with cones of 360 degrees.  I'm gonna take a bath now.
closed 2024-08-26 by yorikvanhavre