FreeCAD/FreeCAD #15600: PartDesign: Sub-Shape Binder Problem

Pendelf, 2024-07-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Creating body1 in PartDesign
Create body2 based on Sub-Shape Binder
Make a hole in body2
Activate body1 and an error appears when trying to create a Sub-Shape Binder of a hole in body2

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.24
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * OpenTheme 2024.7.3
  * PieMenu 1.7.5
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

PartDesign

### Anything else?

https://youtu.be/Byram3zCdQc
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-25:
Could you please provide an example file?

Pendelf, 2024-07-25:
[Sub_problem.zip](https://github.com/user-attachments/files/16371585/Sub_problem.zip)

In version 0.21 the behavior is the same

maxwxyz, 2024-07-25:
I can see. You are trying to reference back a feature which is based on a feature of the other body. Changing that would result in a different shapebinder for the base of Body001 so this is a cyclic dependency. Not sure if this could be solved like this.
Workaround for your file would be to create the sketch001 for the pocket in the other body and create the shapebinder from this.

Pendelf, 2024-07-26:
Everything works fine in FreeCAD Link Branch

https://youtu.be/5Y-bbEQug80

[image]

[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 2024.407.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.8, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.24
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * OpenTheme 2024.7.3
  * PieMenu 1.7.5
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.23
[/code]

maxwxyz, 2024-07-26:
@realthunder could you help to narrow this down?

Syres916, 2024-07-26:
With all due respect to @realthunder's coding skills which are way superior to mine but I think the FreeCAD core cyclic check is correct and should refuse that modelling procedure for good reason. That's not to mention it's just bad modelling practice, far better to have a master sketch in the root of the document if you want to use multiple bodies with the same geometry re-used via CarbonCopy but that's IMHO.

Pendelf, 2024-07-27:
This is probably bad practice from a modeling perspective, but in the other 5 cads I use this can be done

gusgibbon, 2024-09-22:
> With all due respect to @realthunder's coding skills which are way superior to mine but I think the FreeCAD core cyclic check is correct and should refuse that modelling procedure for good reason. [...]

I would disagree that this is bad modeling practice.

Lets say that you have two bodies, body 1 with some initial dimensions and body 2 which is created from the dimensions contained in body 1, and which additionally has some secondary geometry that needs to be reflected back to body 1 (this action causes the cyclic dependency). While yes, you could recreate this secondary geometry again in body 1 and never reference body 2 from body 1, sometimes the secondary geometry is derived though so many steps that this recreation becomes extremely time consuming.

I'm not very familiar with the inner workings of FreeCAD but I believe FreeCAD's sub-shape binder is dependent on the entire body instead of a particular feature within that body. If so then a potential fix could be making the sub-shape binder dependent on features, which would resolve the circular dependency as the second binder could never interfere with the first.
[image]

MikePryadko, 2025-05-29:
## A simpler example
1. create `bodyA1` in `partA`
2. create `bodyB1` in `partB` via `Shape binder` + `Datum pane`
3. create `bodyA2` in `partA`
4. try to create `Binder` in `bodyA2` from face of `bodyB1`
5. get error <mark>\<Exception\> ShapeBinder.cpp(1008): Cyclic reference to shape_binder_cycling_ref_fail_simple#Part001</mark>

## Demonstration

https://github.com/user-attachments/assets/af4d9ba5-26e7-457f-81b8-5806e8338134

Project file: [shape-binder_cycling-ref-fail-simple.FCStd.zip](https://github.com/user-attachments/files/20509424/shape-binder_cycling-ref-fail-simple.FCStd.zip)

## Comment
Looks like "cyclic reference checking" works on Part-level, but I don't understand for what 😏  I think it's **overinsurance** and it should be changed 😌

cyril42e, 2026-05-08:
I believe it is related to #11003 ? In the comments a manual workaround is suggested, to do precisely what @gusgibbon proposed, ie making the binder reference a feature instead of the body (or more accurately, reference a feature not through the body). I guess it also means that the fix is easier than anticipated, if the binder is already able to reference a feature, and it just needs to be configured differently ?