FreeCAD/FreeCAD #15602: Dark theme: Low contrast and no selection indicator in auto hide menu

dapetcu21, 2024-07-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The dark icons are almost impossible to see over the dark background. Also, there is no indication of which of the options is currently selected while the menu is open. It took me a while to figure it out that it is a single selection and not a "bitmask" with a toggle for each of those options.

[image]

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dapetcu21, 2024-07-24:
Here's an example (from Safari, but this is a common pattern on macOS) of what I mean by having a visible selection:
[image]

Syres916, 2024-07-25:
@MisterMakerNL just for info, over the weekend I was going to look into how difficult it would be to implement a light and lighter version of the four icons without actually creating any new ones like the line patterns in Tech Draw:

[image]

So the three icons not selected would be (low) light and the selected be lighter.

The code I'm looking at making the on-the-fly brightness changes to is https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/OverlayWidgets.cpp#L891

I've already made blue versions for Behave-dark but I know that's not a sustainable way to proceed for the core themes.

MisterMakerNL, 2024-07-26:
The icons are already there but I think there not set.
I am not sure why, tho.
Having any icon switched automatically is always a preferred solution tho. So if you can get that too work, then that would be awesome!

Syres916, 2024-07-26:
I didn't realise the overlay icons are only 12x12 pixels so that makes the masking procedure nonviable, needs at least 48x48 pixels to stop them becoming blurry. So in order to get both themes working with regard to the user being able to see which mode they are in, some additional icons were needed.

After lots of testing and various greys (as us Brits spell correctly), I've got a dev branch which I'll squash into a final branch once all the Lint complaints (no doubt) have been sorted https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Overlay_Mode_Enhancement_dev

PR #15635 (2024-08-12): [Gui] Overlay hide mode improve contrast in both themes...
...and ensure selected mode stands out from the deactivated modes.

Fixes #15602

[image]

[image]
closed 2024-08-12 by chennes