FreeCAD/FreeCAD #15603: Stylesheet: Rename text input fields cut off text

maxwxyz, 2024-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Renaming tree view elements cut off text with the default themes (light/dark). Classic theme is ok. This also happens in other views where text is entered (property panel, task panel,..)
macOS: 
[image]

Windows:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jag45, 2024-07-26:
Hi
I noticed the same thing in the parameter editor as here
[image]

I have managed to get round it by making the following change to the dark.qss file

[image]

Add the 0 into line 

[image]

[image]

Don't know if this is correct way to do it but it seems to work.
Jon

MisterMakerNL, 2024-07-27:
Hey Jag45
It makes no sense tho, you removed padding from left and right and only do it on top and bottom.
What you are saying is you added padding by removing padding.
To be honest QT doesn't always make sense. ;)

jag45, 2024-07-28:
Yeah, it does seem strange. What it is doing is the following,

QAbstractItemView QLineEdit {
/*JON padding: 2px; JON*/      This is setting the padding at 2pixels all
round (top, bottom and sides) causing it to shrink the content.
  padding: 0 2px;                       This is now setting the padding to
be 0 pixels top and bottom and 2pixels at the sides.
/*JON ADDED NEXT LINE */
  margin: -2px 0;                       This makes it look better by
setting the margin 2 pixels larger top and bottom thus giving a slight
space top and bottom
}                                                  like this.
[image: image.png]     It doesn't seem to make any difference if you make
it larger than two pixels, there must be some constraint on size elsewhere.
                                                    but you can make the
side padding smaller if you want to.

The box model is explained quite well here:-
https://doc.qt.io/qt-6/stylesheet-customizing.html#the-box-model
and also in examples for padding and border etc.

closed 2024-07-29 by chennes