FreeCAD/FreeCAD #15605: Stylesheet: White borders and white arrows on property view in overlay mode

maxwxyz, 2024-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With the default stylesheets in transparent overlay mode, the property panel has a white border on the right side, and also white blocks at the left of every indented row.
Also, the white arrows on the left on expandable elements are not legible.
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-03:
This is still the case in today's Windows weekly.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

maxwxyz, 2025-07-12:
@MisterMakerNL was this fixed in your pr?