FreeCAD/FreeCAD #15606: Stylesheet: Material dialog colors

maxwxyz, 2024-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The previously white material task view and material editor now have stil white text on a light grey background.

Materials selector, different colors for background and text in no selection, preselection (hover) and selection:
[image]

Materials task dialog:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-03:
I don't see this in today's Windows weekly but it would be better to recheck because I haven't noticed it before either.

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)

maxwxyz, 2024-08-03:
It's fixed

closed 2024-08-03 by maxwxyz