FreeCAD/FreeCAD #15609: Stylesheet: Light theme tree view has dark shadow and appears blurry

maxwxyz, 2024-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With the FreeCAD light theme and overlay with transparency enabled renders a darker than white border around the tree view items which makes them appear blurry and harder to read. The "shadow" should be full white for better contrast.
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-03:
I agree. The tree view still appears blurry in this mode.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

Syres916, 2024-08-11:
This is the result of the changes in this branch https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:FCLight_Improvements

[image]

MisterMakerNL, 2024-08-11:
Hehe you just disabled the effect, but how is it when some part is behind it?

MisterMakerNL, 2024-08-12:
I made this worse. Because I fixed it not showing disabled features.
And it is way worse when the text is gray.
woopsie

MisterMakerNL, 2024-08-12:
Are we going the Opentheme way and just adding a background?

maxwxyz, 2024-08-13:
I think just a full white "shadow" will fix this. It works great in dark.

closed 2024-08-26 by chennes