FreeCAD/FreeCAD #15613: Stylesheet: Dark tab colors not ideal

maxwxyz, 2024-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With FreeCAD Dark theme the tab color could be improved:

Preselection of the tabs renders them not legible (white text on light grey background):
[image]

The active tab is lighter than the inactive tabs. On dark themes, the dark tabs appear "pressed" and therefore active. IMO the colors should be switched:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-03:
The first part of this seems to be fixed in today's Windows weekly.

maxwxyz, 2025-07-12:
@MisterMakerNL and this too?