FreeCAD/FreeCAD #15615: Fusion in part WB failed in 0.22.0dev.38314 but not 0.21.2.

kevinsmia1939, 2024-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Applying Fusion in part workbench between 2 objects failed in weekly build 0.22.0dev.38314 appimage, but latest stable release perform boolean successfully.

Remove the .txt extension to open it.
[boolean_fail.FCStd.txt](https://github.com/user-attachments/files/16376899/boolean_fail.FCStd.txt)

FreeCAD 0.21.2
[image]

FreeCAD 0.22.0dev.38314
[image]

### Full version info

```shell
OS: openSUSE Tumbleweed (KDE/plasma6)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git) AppImage
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * QuickMeasure 2022.10.28

Stable version
OS: KDE Flatpak runtime (KDE/plasma6)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.61
  * CfdOF 1.25.1
  * Assembly4 0.50.14
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-25:
Is this the same issue as https://github.com/FreeCAD/FreeCAD/issues/15599

krushia, 2024-07-25:
> Is this the same issue as #15599

Yep. This is working with OCC 7.6.3.

kevinsmia1939, 2024-07-25:
> Is this the same issue as #15599

Unlike the issue mentioned, I could not see any error or warning in the report view.
Here is all the log when FreeCAD opened and after boolean fusion.
```
14:15:09  Init: Activating default workbench PartDesignWorkbench
14:15:09  Loading Sketcher module... done
14:15:09  Loading PartDesign module... done
14:15:09  Loading GUI of Material module... done
14:15:09  Loading GUI of Part module... done
14:15:09  Loading GUI of Sketcher module... done
14:15:09  Loading GUI of PartDesign module... done
14:15:09  Init: Showing main window
14:15:09  Main window restored
14:15:09  Show main window
14:15:09  Toolbars restored
14:15:09  Couldn't connect to spacenav daemon. Please ignore if you don't have a spacemouse.
14:15:09  Init: Entering event loop
14:15:09  Init: Executing event loop...
14:15:09  Init: Processing command line files
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:17  	Property 'SpecificCuttingForce' is not described by any model. Ignored
14:15:50  Sketcher::setUpSketch()-T:0.00194991
14:15:50  Sketcher::Solve()-DogLeg-T:0.000173486
14:15:50  Sketcher::setUpSketch()-T:0.000474643
14:15:50  Sketcher::Solve()-DogLeg-T:1.7042e-05
```

Syres916, 2024-07-25:
Works with OCC 7.8.1 as well.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38331 (Git)
Build type: Release
Branch: main
Hash: e4e17cf398294c75dc5a09dac3ee6b2bb778fa52
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * A2plus 0.4.65
  * FreeCAD-themes 2024.7.24
  * CfdOF 1.27.4

```

krushia, 2024-07-25:
> Works with OCC 7.8.1 as well.

That makes this different from #15599, which is still broken on 7.8.1.

oursland, 2024-08-17:
Current weekly builds use `occt==7.7.2`.  I can look to see if we can specify a different version on package install.

oursland, 2024-08-17:
The weekly builds can build with `occt==7.6.3`, but there are versioning conflicts with `7.7.0`, `7.7.1`, and `7.8.1`.

We can pin the weekly builds to a specific version of `occt`, but other packages (ppa, snap, appimage, etc) may still build against other versions of `occt` that could exhibit this issue.

bgbsww, 2024-08-17:
I am unable to replicate the bug as stated - the boolean does render successfully on a build against 7.7.2.

It does, however, take a very large amount of time - multiple minutes to complete the rendering of the screw threads.

I'm not sure if this is because we changed something in the last few days and it is now a little bit better, or if there's something else going on here.

OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38470 (Git)
Build type: Debug
Branch: bgbsww-toponamingFuseRefineElement
Hash: d6ce297893d9d5e887c653e08ef676fb9be7b50b
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD_BearingsWB 0.0.2
  * sheetmetal 0.4.18
  * DynamicData 2.62.0
  * BIM021 2021.12.0
  * parts_library

krushia, 2024-08-20:
Still broken here. Tried 7.7.2 and 7.7.0. Ignore messages in screenshot, they're for something else.

[image]

OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git) AppImage
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.42
  * A2plus 0.4.65
  * DynamicData 2.62.0
  * Assembly4 0.50.15
  * Defeaturing 1.2.2
  * dodo 1.0.1 (Disabled)
  * boltsfc 2022.11.5
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * Design456 0.0.1
  * fasteners 0.5.25
  * SearchBar 1.0.2
  * DesignSPHysics 0.7.0(15-09-2023)
  * cadquery_module
  * ose-piping (Disabled)
  * Nodes 0.1.36
  * 3D_Printing_Tools
  * Pyramids-and-Polyhedrons
  * workfeature
  * MnesarcoUtils 0.2.5
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * CommandPanel
  * CubeMenu
  * MeshRemodel 1.9.23

bgbsww, 2024-08-24:
aa
[image]

Absolutely cannot replicate.  Is there something else about your setup?  Maybe not an OCCT bug, but something else?

Daily:

OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)

Weekly:

OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git) AppImage
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)

Locally built against 7.7.2:

OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38470 (Git)
Build type: Debug
Branch: bgbsww-toponamingFuseRefineElement
Hash: d6ce297893d9d5e887c653e08ef676fb9be7b50b
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.7.2
Locale: English/United States (en_US)

kevinsmia1939, 2024-08-24:
> ![image](https://private-user-images.githubusercontent.com/120601209/361152756-c42faaf2-907d-48d6-bd00-1b8c0cbaafd5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ1Mzg5ODUsIm5iZiI6MTcyNDUzODY4NSwicGF0aCI6Ii8xMjA2MDEyMDkvMzYxMTUyNzU2LWM0MmZhYWYyLTkwN2QtNDhkNi1iZDAwLTFiOGMwY2JhYWZkNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyNFQyMjMxMjVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNTMxZGUyMjJiOTQxNDlkZTVlNzlkNmE0MWU4MGFmOWFiZjMwMjBkYzUyMWQyZTlmNTMxMGUwMDEyNTZlNGFjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.4Yk8zAq0qCtAigB3Orz6Q-5aKgixQm6i_9cFZQ5994o)
> 
> Absolutely cannot replicate. Is there something else about your setup? Maybe not an OCCT bug, but something else?
> 
> Daily:
> 
> OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu) Word size of FreeCAD: 64-bit Version: 0.22.0dev.38506 (Git) Build type: Release Branch: main Hash: [ebf0248](https://github.com/FreeCAD/FreeCAD/commit/ebf02483af225b502fccfa7111499dfde33999fd) Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1 Locale: English/United States (en_US)
> 
> Weekly:
> 
> OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu) Word size of FreeCAD: 64-bit Version: 0.22.0dev.38506 (Git) AppImage Build type: Release Branch: main Hash: [ebf0248](https://github.com/FreeCAD/FreeCAD/commit/ebf02483af225b502fccfa7111499dfde33999fd) Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2 Locale: English/United States (en_US)
> 
> Locally built against 7.7.2:
> 
> OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu) Word size of FreeCAD: 64-bit Version: 0.22.0dev.38470 (Git) Build type: Debug Branch: bgbsww-toponamingFuseRefineElement Hash: [d6ce297](https://github.com/FreeCAD/FreeCAD/commit/d6ce297893d9d5e887c653e08ef676fb9be7b50b) Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.7.2 Locale: English/United States (en_US)

I tried the weekly Appimage build and can still replicate the bug.
```
OS: openSUSE Tumbleweed (KDE/plasma6)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git) AppImage
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * QuickMeasure 2022.10.28
```

bgbsww, 2024-08-24:
I'm going to unassign, because in my world, there is no bug here.  If someone can give more conclusive evidence that this actually happens, I'll take a look, but I've spent enough time trying everything, and it simply does not fail.

Maybe another developer can find a way to see it fail.

maxwxyz, 2024-08-25:
@FlachyJoe you just created a helix PR, could you look into this issue?

FlachyJoe, 2024-08-25:
@maxwxyz I can not do more tests than bgbsww, if we can't replicate we can't diagnostic.

krushia, 2024-08-26:
The fusion isn't in the file, you need to create it. Maybe that's the confusion?

bgbsww, 2024-08-27:
Okay, so the fusion is failing because the 21.15 constraint on the sketch that gets revolved means that not all of the shape components of the helix are intersecting with the cylinder, and that results in multiple solids which throws an error.

Changing the 21.15 to 21.16 causes the boolean fuse to start working.  Actually, so does 21.151 mm - anything to avoid the edge condition.

While this would ideally be solved in the engine ( and apparently has been in 7.8.1 ), there isn't really a viable workaround for the FC interface that I can come up with.

I know from other threads that we're likely to keep working on variations of the helix implementation to try to make them work better and with better performance, but cases like this that rely on exact clearances between a helix and another shape are likely to continue to be brittle.

If I'm missing something, let me know.

bgbsww, 2024-08-27:
Also, if we accept my diagnosis, this this is an unfixable, unworkaroundable in code bug in OCCT 7.7.2 only; good in 7.6.3 and 7.8.1.  I think that means we drop the blocker, and leave the bug so that others can find this if they hit it.

The practical advice is to make sure your helixes overlap (not just touch) with the objects they are booleaned with - a little nub in the helix sketch pointing inward should act as another possible fix here if you need to cylinder dia to be exact.

kevinsmia1939, 2024-08-28:
@bgbsww 
If it is working in newer version of occt then can we ship new occt with upcoming freecad release?

bgbsww, 2024-08-28:
@kevinsmia1939 That is not my area of expertise, but as I understand it, we can't currently do that, we rely on the underlying platform for some releases.  I can say that the latest ubuntu is moving from 7.7.2 to 7.8.1 so OS updates should solve it, and I think the appimage type builds include their own dependencies, but the build experts would know all the details.  I can only really speak from the is-there-a-possible-workaround-for-this-underlying-bug perspective here.

chennes, 2024-08-31:
I've dropped the Blocker label on this one based on @bgbsww's comments. OCCT 7.7 turns out to have several pretty strange bugs in it, by and large we should endeavor not to ship anything linked against it (but that's really a packaging problem).

maxwxyz, 2024-08-31:
@adrianinsaval ^ FYI

AIRCAP, 2024-10-09:
i was hoping #17119 would fix it, but it seems to have no effect on this particular case.

AIRCAP, 2024-10-17:
ok now its fixed - fixed in #17119

PR #17119 (2024-10-21): Fix boolean operations
Fix #17159
Fix #17150
Fix #17085
Fix #16081 
Fix #15825
Fix #15615
can alleviate #11083 ( fixes with nonstandard setting for boolean fuzzy > 1000 - but that's too high to cut out very delicate parts from very large blocks )
Fix #9666
Allows workaround for #17338 

OCCT has issues with boolean operations failing when parts are touching but not intersecting. This includes but is not limited to the PartDesign helix , which is notorious for producing results that do not pass BOP tests or even fail completely.

The suspected problem is too narrow precision. Part precision sho [... cut]