FreeCAD/FreeCAD #15616: Stylesheet: FreeCAD Dark "more" arrows on toolbar not visible

maxwxyz, 2024-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The "more" arrows in the toolbar are not visible until hovered (dark icon on dark background):
[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-07-26:
They look a little unusual/low-res on the Mac, too:

[image]

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * Curves 0.6.42

MisterMakerNL, 2024-07-27:
These icons aren't themable, i can only change the background.
Which I did long ago but because of a QT 6 bug it got removed.

MisterMakerNL, 2024-07-27:
https://forum.qt.io/topic/119988/qtoolbar-extension-button-stylesheet/2
I'll change the background to something light for now./

closed 2024-07-29 by chennes