FreeCAD/FreeCAD #15620: Stylesheet: FreeCAD Light - Selection background color slightly too dark

maxwxyz, 2024-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

IMO there is not enough contrast between the dark blue background color and the black text color of selected and preselected items in FreeCAD Light. 
This is visible in the menu, preferences, input fields,...
[image]
[image]
[image]
[image]

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.22
  * Curves 0.6.41
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-07-25:
These are user assigned colors. 
I can set them in the theme, but this would overwrite any user settings.
[image]

maxwxyz, 2024-07-25:
I thought they could be changed with a theme, like also the background color. They usually are also defined by the user but are overwritten.

MisterMakerNL, 2024-07-25:
That is a valid point and you probably also have to change them since your colors set in light will probably not work in dark.
Do we have aby suggestion which colors?
Freecad logo colors? @FreeCAD/design-working-group ?

FEA-eng, 2024-08-03:
This got fixed by #15586, I tested it in the newest Windows weekly.

closed 2024-08-03 by luzpaz