FreeCAD/FreeCAD #15625: Crash when opening Help menu without an internet connection

hooperbloob, 2024-07-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Like the title says, clicking on the (?) Help menu option causes FreeCAD to crash when I don't have an internet connection.

[image]

Under v0.21 it pops up a dialog:

The Help system of FreeCAD is now handled by the "Help" addon. It can easily be installed via the Addons Manager

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * BIM021 2021.12.0
  * fasteners 0.5.24
  * sheetmetal 0.4.12
  * Assembly4 0.50.15
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-07-26:
No crash here.

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38323 (Git)
Build type: Debug
Branch: main
Hash: 6a5207f7a8894618f9d693026e15dc5b47c1952e
Python 3.10.12, Qt 6.7.2, Coin 4.0.0, Vtk , OCC 7.5.1
Locale: English/United States (en_US)
```

Can you provide the crash report?

neotwix, 2024-07-26:
I have the crash with the last weekly buid.
Launch Freecad and Just clic Help

 ./FreeCAD_weekly-builds-38314-conda-Linux-x86_64-py311.AppImage 
FreeCAD 0.22.0, Libs: 0.22.0devR38314 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
Could not find QtWebEngineProcess
/tmp/.mount_FreeCACguQOk/AppRun : ligne 41 : 3306804 Abandon                 (core dumped) "${MAIN}" "$@"

neotwix, 2024-07-26:
@xtemp09 I think you build it with QT6. the weekly build is QT5. and QtWebEngineProcess seem missing.

chennes, 2024-08-11:
There should be no QtWebEngine process, nor a required reference to it. Is this one of the Linux Conda builds?

chennes, 2024-08-11:
For the record, I get no crash with:
```
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38434 (Git)
Build type: Unknown
Hash: 99375384c292c8203e5b2dd9e059f29b63318b86
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.3.0
Locale: English/United States (en_US)
```
What options do you all have selected in Preferences->General->Help?

hooperbloob, 2024-08-17:
With the latest build, the main window disappears when Help is clicked sans internet but the app still seems to be running. 
Clicking on the app in the toolbar brings the window back up with the current sketch seemingly intact.

OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git)
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * BIM021 2021.12.0
  * fasteners 0.5.24
  * sheetmetal 0.4.12
  * Assembly4 0.50.15

oursland, 2024-08-17:
I cannot reproduce the issue with the most recently weekly build.  The behavior I get is that the default browser is opened to the page `https://wiki.freecad.org/Online_Help_Startpage`, which cannot load.  FreeCAD is still running.

sliptonic, 2024-08-17:
Can't duplicate on linux (popos)

luzpaz, 2024-08-18:
@sliptonic can you add Full About info, TIA!

luzpaz, 2024-09-09:
@hooperbloob please recheck on latest RC1.0 release, TIA

hooperbloob, 2024-09-09:
The browser comes to the front, complains about not being able to open the website (as expected)
The FreeCAD window sits patiently behind it.

Seems ok now - right on.

luzpaz, 2024-09-09:
@hooperbloob can you paste your [About](https://wiki.freecad.org/About) info for us to reference? TIA

luzpaz, 2024-09-09:
In addition to the About info, can you also throw this link https://wiki.freecad.org/Online_Help_Startpage into your browser and see if it resolves ?

hooperbloob, 2024-09-09:
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * BIM021 2021.12.0
  * fasteners 0.5.24
  * sheetmetal 0.4.12
  * Assembly4 0.50.15

> can you also throw this link https://wiki.freecad.org/Online_Help_Startpage into your browser and see if it resolves ?

The browser opens the page as expected when my machine is connected to the internet, either directly or via the FreeCAD Help menu.

luzpaz, 2024-09-09:
hey @johnocfii would you mind testing to see if this is a macOS related bug ?

JohnOCFII, 2024-09-10:
If not connected to the Internet, I get a Safari browser window with the message that I can't connect to Internet.  If I AM connected to Internet, this page is reached: https://wiki.freecad.org/Std_WhatsThis

FEA-eng, 2025-10-04:
@hooperbloob Can you check if the issue still occurs with the newest weekly build from [here](https://github.com/FreeCAD/FreeCAD/releases)?

FEA-eng, 2025-10-05:
@maxwxyz Could you close this?

closed 2025-10-05 by maxwxyz