FreeCAD/FreeCAD #15627: Assembly: Make first part transparent after selection when creating joints

tibfel, 2024-07-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a feature Request: I have seen this in Solidworks and it saves a lot of time when joining parts together. When I make joints between two parts, the first part often gets in the way of selecting face/edge of the second part. The view must be rotated or the first part must be moved out of the view to access the face/edge of the second part to be mated. If the first part was transparent (i.e. you can select through it), it would eliminate the unnecessary step of rotating view or moving the first part. 

Probably this will not get into 1.0 version, as there is feature freeze, but it would be a nice to have feature in the future. 

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Slovak/Slovakia (sk_SK)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.7.3
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-26:
The regnest looks similar to the approach in Part Design: https://github.com/FreeCAD/FreeCAD/issues/11214

tibfel, 2024-07-26:
@maxwxyz I think that if that feature in Part Design is developed, it can be reused with some modifications in the assembly workbench for my issue as well. The underlying technology looks similar, but it is not the same issue. For me, it is crucial to be able to see through the part and also be able to pick objects behind the already selected part, so it does not get in a way. 

Maybe a high-level workflow would be: after one part has selected edge/face/point, suppress it in the model (without recomputing the joints), but do not hide it, only make it transparent, so I can select the other part through it. Once the joint is successful, make the previously suppressed and transparent part normal (opaque) again.

tibfel, 2024-07-26:
I have created a demo in SolidWorks to show what I am talking about and its usefulness. There is no need to rotate the view or move the rod behind the board to get to the mating face. 
[image]