FreeCAD/FreeCAD #15631: VarSet: Failed to parse expression in a variable set

tibfel, 2024-07-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have created a variable set called Variables with two parameters da and db, they are of type App:ProperyLength
[image]
When I try to use variable in a sketch 

I am not able to click OK and I get error Failed to parse expression in variable set. 
[image]. 

I have tried App:ProperyLength and App:ProperyFloat, but the result is the same. 
here is the file:
[variable sets.zip](https://github.com/user-attachments/files/16390538/variable.sets.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Slovak/Slovakia (sk_SK)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.7.3
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-07-26:
Unlike in spreadsheet cells, you don't need the = in the expression field vitself.

Pressing = in the normal field brings up the expression editor. But from there, just type the expression there without the = and it will work.
[image]

tibfel, 2024-07-26:
Thank you very much. I can confirm that this works :)

closed 2024-07-26 by tibfel