FreeCAD/FreeCAD #15633: Pad added to wrong sketch

psommerfeld, 2024-07-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have two sketches, one called "top", the other called "bottom".
If I pad one, it seems to pad correctly. When I pad the other one, it seems to add this pad over top of the first one again.
I previously deleted some other sketches so maybe TNP issue?

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Generic)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git) AppImage
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Anything else?

[grinder disc cover.zip](https://github.com/user-attachments/files/16392129/grinder.disc.cover.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Pendelf, 2024-07-27:
So you want to create two bodies?

Pendelf, 2024-07-27:
Everything works fine, attached the file. It's just that when the extrude operation is applied, there is no preview, and the extrude was going the other way and it's not visible
[grinder disc cover 1.zip](https://github.com/user-attachments/files/16398335/grinder.disc.cover.1.zip)

wwmayer, 2024-07-27:
Because the 2nd sketch is not attached to a face of the 1st pad it will create a compound of multiple solids. In this case you must select the Body and set its _Allow Compound_ property to True.

maxwxyz, 2024-07-27:
I'll close this issue then. For further help please create a post on the forum.

closed 2024-07-27 by maxwxyz