FreeCAD/FreeCAD #15634: Stylesheet: Tree view has two colored selection for a row

maxwxyz, 2024-07-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using the FreeCAD themes, the tree view displays two colors for the same selected element: light blue and dark blue. 
IMO the light blue on the left of the element is not necessary and could be removed, it is also not visible with preselection.

[image]

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-30:
@MisterMakerNL in your Qt6 build with the latest PR it looks like this: 
[image]

FEA-eng, 2024-08-03:
I don't see this in today's Windows weekly but maybe I have some different settings. Can you recheck?

maxwxyz, 2024-08-03:
Fixed

maxwxyz, 2024-08-04:
Ok, fixed in Windows but still in the latest macOS weekly:
[image]

```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42

```

@MisterMakerNL

MisterMakerNL, 2024-08-04:
IT happens in windows when you un-focus (select anything then click outside the Freecad window.
[image]

maxwxyz, 2024-08-04:
it happens all the time on macOS (I have transparent overlay for the tree view enabled)
Clicking in the 3D view or on a TechDraw page just deselects the item.
https://github.com/FreeCAD/FreeCAD/issues/15659

Selecting and edge in the 3D view selects the sketch in the tree with two colors:
[image]

```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

maxwxyz, 2024-08-06:
This is with classic:
[image]

```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42

```

Syres916, 2024-08-11:
> I think this is some selection item that isn't set in stylesheets but is used, so it used default selection colors.

See https://github.com/FreeCAD/FreeCAD/issues/15609#issuecomment-2282320096 how I split TreeView off in FreeCAD Light.qss

MisterMakerNL, 2024-08-11:
Should be fixed here although I'm sure it caused some other tinier bugs https://github.com/FreeCAD/FreeCAD/pull/15806

Syres916, 2024-08-11:
I cleared out my build folder, recompiled and then applied PR15806 and neither FreeCAD Dark nor FreeCAD Light were fixed with regard to this issue. I updated my branch https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:FCLight_Improvements to include FreeCAD Dark, re-enabled the blurring effect for Light Scheme/Light Background in order to try to get a more acceptable result but what I think is acceptable probably isn't to others. With regard to FreeCAD Classic that's going to take me a while to understand some of the code and basically check for no stylesheet and if so, apply the same basic style settings but it's going to be probably 10 days before I'm going to get some serious time for it.

[image]

MisterMakerNL, 2024-08-11:
The fix was in the overlay light/dark.
I'll check light/light and dark/dark. Thought it wouldn't be in there.

MisterMakerNL, 2024-08-11:
I still have to check the blurry text, but I am trying to do really obvious bugs like the selection color first.
[image]

MisterMakerNL, 2024-08-11:
> I cleared out my build folder, recompiled and then applied PR15806 and neither FreeCAD Dark nor FreeCAD Light were fixed with regard to this issue. I updated my branch [main...Syres916:FreeCAD:FCLight_Improvements](https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:FCLight_Improvements) to include FreeCAD Dark, re-enabled the blurring effect for Light Scheme/Light Background in order to try to get a more acceptable result but what I think is acceptable probably isn't to others. With regard to FreeCAD Classic that's going to take me a while to understand some of the code and basically check for no stylesheet and if so, apply the same basic style settings but it's going to be probably 10 days before I'm going to get some serious time for it.
> 
> ![Screenshot from 2024-08-11 13-00-24](https://private-user-images.githubusercontent.com/46537884/356872626-d9983b88-cd54-4d74-9114-207720086296.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMzOTA3OTcsIm5iZiI6MTcyMzM5MDQ5NywicGF0aCI6Ii80NjUzNzg4NC8zNTY4NzI2MjYtZDk5ODNiODgtY2Q1NC00ZDc0LTkxMTQtMjA3NzIwMDg2Mjk2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODExVDE1MzQ1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdiMmE2ZjVlMzZmZTdlYTZkYTMzZjBkYmExMGUxYjg1NTQ3MTFkN2U4Mzk5MTEyYmNlOTNkMjIwYjJhNDA1MTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.e11LYBl8ThcvTgVLb0NpzVBZ5syA8lkXMQn67cjOfuw)
[image]

closed 2024-08-12 by chennes