FreeCAD/FreeCAD #15638: Open SSL has vulnerabilities and Phyton is nearly EOL

ThomasKur, 2024-07-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Our vulnerability scanning shows that in the last public version 0.21.2 are multiple vulnerable libraries used: 

Software/Component | Version | Evidence | Weaknesses
-- | -- | -- | --
Python | 3.8.17.0   (upcoming EOS version) | C:\Program Files\FreeCAD 0.21\bin\py.exe | 4
Openssl | 3.0.9.0 | c:\program files\freecad 0.21\bin\libcrypto-3-x64.dll   c:\program files\freecad 0.21\bin\libssl-3-x64.dll | 7
Libwebp | 1.2.0.0 | c:\program files\freecad 0.21\bin\libwebp.dll | 2

Would it not be possible to allow updating libraries separately or more often than releasing a new FreeCAD version? So, for example having these libraries in separate folders and allow us updating the files by exchanging them? 

The above is just a current list and there are potentially others with vulnerabilities.

### Full version info

```shell
0.21.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-07-27:
0.22-dev appears to be built against Python 3.11.9, which has three more years of security releases, so this is obviously on the radar, as are all the issues regarding interpreter locks if you look at the issues etc.

Not sure about libwebp but re: SSL, in 0.22-dev,

```
>>> import ssl
>>> ssl.OPENSSL_VERSION
'OpenSSL 3.3.1 4 Jun 2024'
```

I'm just an end user, but I can't see any particular ecosystem benefit in a version where end users can have different installable versions of python, given that the source of FreeCAD is available to build for those who would be tinkering at that level.

maxwxyz, 2024-07-27:
@adrianinsaval is this an issue or already taken care of for 1.0?

adrianinsaval, 2024-11-05:
> @adrianinsaval is this an issue or already taken care of for 1.0?

yes we will not be using the old libpack with these dependencies anymore

closed 2024-11-05 by adrianinsaval