FreeCAD/FreeCAD #15641: 'Remove objects and their children' tool in the OpenSCAD workbench causes a crash sometimes.

TadPath, 2024-07-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to remove a model and all its children by selecting the model then click on the 'Remove objects and their children' tool in the OpenSCAD workbench, this causes the whole program to crash out completely without warning - but only sometimes.

### Full version info

```shell
When trying to remove a model and all its children by selecting the model then click on the 'Remove objects and their children' tool in the OpenSCAD workbench this causes the whole program to crash out completely without warning - but only sometimes.

I do not have OpenSCAD installed (and never have).

This remove feature worked well and reliably in previous versions and I never experienced a crash with it before. It only started happening when I upgraded to the following FreeCAD:

[code]
OS: Arch Linux (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * fasteners 0.4.15
  * OpticsWorkbench 1.0.9
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-27:
Could you try this in the current dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Can you provide your demo file and steps how to reproduce, you've mentioned it only happens sometimes.

TadPath, 2024-07-27:
Hello,

I downloaded the file: "[FreeCAD_weekly-builds-38334-conda-Linux-x86_64-py311.AppImage](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-38334-conda-Linux-x86_64-py311.AppImage)"

When I run this dev version the problem does not occur - no crashes. I tried several model from different files. Specifically, see the attached file CrashTest.FCStd.

Load file
Ensure tree view is visible
Select OpenSCAD workbnech (get warnings about OpenSCAD not installed - this is normal)
Left click the first model in the tree (called "M3_Adjustment_ring_Kohler")
Left click the 'Remove objects and their children' icon
The object and its dependents are removed, no problem
Left click the second model in the tree and repeat - the model is removed, no problem
 
I confirm that when I do the above procedure with the current release version of FreeCAD it does still crash when deleting the models in this way - sometimes it will crash deleting the very first model but other times it will allow one or two of the models to be deleted like this and then crash when I try to delete the next one on the list.

So it seems that you have fixed this specific issue in the above dev version.

However, while testing I noticed a different problem in your dev version:

If you open the supplied Crashtest.FCStd, select one of the measurements in the tree view  - for example left click on the one labelled "Distance: 5.93 mm" - then press the 'Del' button on the keyboard, it opens up the tree of all the parts in my models and shows lots of errors in the models (errors that did not exist before). If you do the same thing with the current release version then it simply deletes the measurement object, as expected, with no problems.
This is a separate issue that seems to have crept into the dev version - not the same issue that is the main theme of this email.

Thanks for your help.

P
-----
[crashtest.zip](https://github.com/user-attachments/files/16400329/crashtest.zip)

maxwxyz, 2024-07-27:
Thanks for the follow up! I'll close this issue then as fixed in the next version.

closed 2024-07-27 by maxwxyz