FreeCAD/FreeCAD #15642: Material: Part_Compound does not display colors of source objects

Roy-043, 2024-07-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a Part_Box.
2. Create a Part_Cylinder.
3. Change the DiffuseColor of one of the objects.
4. Select both and create a Part_Compound.
5. Result: The compound has a single color.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-27:
@davesrocketshop FYI

Roy-043, 2024-08-07:
The same problem occurs for Part Booleans.

maxwxyz, 2024-08-17:
@FreeCAD/design-working-group what should be the correct behavior in this case?

This is the related FR for PartDesign: https://github.com/FreeCAD/FreeCAD/issues/11210

obelisk79, 2024-08-17:
There are probably a couple of ways this could work, and i don't think there is a clear logical 'best practice' in this. I would speculate that it would be worth just the final object assume the color settings of the first object selected for the boolean.

pierreporte, 2024-08-17:
I think that both ways are useful depending on the intent and workflow. Ideally there should be an option for each feature (I mean not only for boolean operations) to allow the user to use the color of the main object or body, or to keep its color but it feels too much.