FreeCAD/FreeCAD #15644: FEM FemMesh2Mesh tool doesn't work with quad elements

FEA-eng, 2024-07-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The [FEM FemMesh2Mesh](https://wiki.freecad.org/FEM_FemMesh2Mesh) tool can't handle quadrilateral elements.
Steps to reproduce:
1. Create Part Plane
2. Mesh it with Gmsh in FEM, set Recombine All to true. Effect:

[image]

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git)
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

fandaL, 2024-07-27:
For volumes, there is a list of possible element types `faceDef`, but for the shells, there is only one type with 3 nodes.
Following the workflow from volumes, this seems to fix the issue:
https://github.com/fandaL/FreeCAD/commit/54cd7895bcdf294d9273d1416225b3c91de82e68

FEA-eng, 2024-07-27:
@fandaL Thanks, it's indeed working with this change. Can you submit a PR?

closed 2024-07-29 by chennes