FreeCAD/FreeCAD #15645: unsolvable sketches, mispositioned and immovable points, vanishing external geometry,

UltraBlackLinux, 2024-07-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hey there,
after having spent a few hours on a project I'm running into issues that are grinding all progress to a halt.
I'm now encountering sketches that were previously fine that now become unsolvable for no apparent reason. 
For example the `swissonic-audio-2-plate/holding lips/Sketch013`:
Points and lines have moved where they shouldn't be. Some lines have jumped to the opposite side of their parent line (the left and right box). Trying to fix those I get errors about immovable points. The problem doesn't seem solvable just by deleting singular points. Almost all connected structures are immovable, rendering the whole sketch useless.
Deleting the construction that connects the Z to the X axis in the sketch ends up deleting almost every remaining construction geometry alongside the external geometry it was initially connected to. That top left point is also refusing to be re-connected like it previously was: `Redundant constraint is not an autoconstraint. No autoconstraints or additional constraints were added. Please report!`
Exiting and re-opening the sketch freecad discovered that each and every single constraint is now redundant. 
Other notable sketch is `panels/back panel/cable hole/sketch010` which also clears its external geometry on interaction.

What do I do now? Is this salvageable? I don't think I can afford to delete and re-create the affected sketches because that would mean rebuilding almost the entire project from scratch. 

[deskrack.FCStd.zip](https://github.com/user-attachments/files/16400773/deskrack.FCStd.zip)

### Full version info

```shell
OS: EndeavourOS (Hyprland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Installed mods: 
  * OpenTheme 2024.7.3
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-28:
You need one additional constraint in that sketch to dimension the width:
[image]
Please consult the forum for help requests.
If there seems to be a bug feel free to reopen this issue and rephrase to be reproducible and actionable.

UltraBlackLinux, 2024-07-28:
The problem I described previously definitely sounded like a whole lot of bugs to me.

The part that puzzles me is that after having rebooted my system to write this after a night, almost all the problems I'd mentioned have now vanished.

Maybe there's something wrong with the tempfiles freecda uses?

Regarding your suggestion to just constrain the retaining lips' depth - I cannot do that since that would overconstrain the whole sketch.

luzpaz, 2024-07-28:
@UltraBlackLinux just to note, main branch will break from time to time. It's good to find a dev build that you can keep as a reliable build simultaneous to test bleeding edge builds. You can also try using the Flatpak or Snap packages to make sure it's not a packaging issue.

luzpaz, 2024-09-10:
@UltraBlackLinux have you been able to reproduce any of these issues on the latest 1.0RC or weekly ?

UltraBlackLinux, 2024-09-11:
I in fact cannot reproduce the issue at this time, so consider this fixed. I'll report back if I run into it again :)

closed 2024-09-11 by luzpaz