FreeCAD/FreeCAD #15646: G85 is issued (linuxcnc) with a feed rate

edmarwjr, 2024-07-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

G85 is issued (linuxcnc) with a feed rate

### Full version info

```shell
This fixed the problem for me:

diff --git a/src/Mod/CAM/Path/Geom.py b/src/Mod/CAM/Path/Geom.py
index eeb00973dd..e39a3be576 100644
--- a/src/Mod/CAM/Path/Geom.py
+++ b/src/Mod/CAM/Path/Geom.py
@@ -86,7 +86,7 @@ CmdMoveRapid = ["G0", "G00"]
 CmdMoveStraight = ["G1", "G01"]
 CmdMoveCW = ["G2", "G02"]
 CmdMoveCCW = ["G3", "G03"]
-CmdMoveDrill = ["G73", "G81", "G82", "G83"]
+CmdMoveDrill = ["G73", "G81", "G82", "G83", "G85"]
 CmdMoveArc = CmdMoveCW + CmdMoveCCW
 CmdMove = CmdMoveStraight + CmdMoveArc + CmdMoveDrill
 CmdMoveAll = CmdMove + CmdMoveRapid
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-27:
@edmarwjr please post your full [About](https://wiki.freecad.org/About) info in the appropriate place in the ticket. TIA

edmarwjr, 2024-07-27:
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38340 (Git)
Build type: Unknown
Branch: main
Hash: 58b3d0217f92c4dbf2e1bc4c42a31afcf4564c24
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * opencamlib
  * Render 2023.12.28

PR #15922 (2024-08-19): Add Feedrate to G85 command moves
#fixes #15646
commit 4ec94e67 (2024-08-19): Add Feedrate to G85 command moves
#fixes #15646
closed 2024-08-19 by WandererFan