FreeCAD/FreeCAD #15650: Can't create no Datum plane nor D. Line nor D. Point, External Geometry does not project intersecting point to actual sketch plane

Carcaman, 2024-07-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Part Design create Body, create sketch, draw a rectangle, extrude it, select a side of it, try to create Datum Plane and nothing happens. Same for Datum line or point.
Othre issue: create sketch in XZ plane, draw a line in an angle that crosses Z axis, close sketch, create another Sketch in YZ plane, use external geometry and extremes points are projected but not the one where line intersect de actual Sketch 

### Full version info

```shell
OS: Windows 10
Freecad 0.22 release 38334
Python 3.11.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-28:
I cannot reproduce. Please provide full about info and a demo file.
Your second point is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/12670

Carcaman, 2024-07-28:
[image]

Carcaman, 2024-07-28:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git)
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * A2plus 0.4.61
  * Assembly4 0.50.13
  * boltsfc 2022.11.5
  * CurvedShapes 1.0.5
  * Curves 0.6.18
  * Dracula 0.0.5
  * fasteners 0.5.22
  * freecad.gears 1.2.0
  * Glass (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * Part-o-magic 1.1.0
  * ProDarkThemePreferencePack 1.0.0
  * QuickMeasure 2022.10.28
  * Render 2023.8.18
  * Silk 0.1.5

Carcaman, 2024-07-28:
> I cannot reproduce. Please provide full about info and a demo file. Your second point is a duplicate of #12670

Sorry, I don't know how to upload a demo file, my first time here.

Syres916, 2024-07-28:
All three are created as expected using:

```
OS: EndeavourOS (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38341 (Git)
Build type: Release
Branch: main
Hash: 00990cb1e193bb5443f8288d7932d5a266f1ec21
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.7.3
```

[image]

Carcaman, 2024-07-28:
Thank you Syres916!!, what could be causing that beheavior? "View provider not attached" "Access violation" "Freecad exception"

Carcaman, 2024-07-28:
If a save the file and I open it in 0.21.2 it seems to be ok, but opening in 0.22 Dev 38334 or previous datums desapear

Syres916, 2024-07-29:
Rather me wasting time doing a git bisect between Friday evening and Sunday morning, can we just agree that if it's fixed in tonight/tomorrow's weekly build then this can be closed?

Carcaman, 2024-07-29:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
 I've just downloaded last version and get same issue. Erased, and back up, original User/AppData/Roaming/ Freecad directory for fresh "install" and as I said, sadly, same behavor.

Syres916, 2024-07-30:
Thanks for the feedback, I'm glad I didn't waste my time then as I wouldn't have found anything. I've just built a brand new Linux Mint 22 vm and it worked without any need to recompute, just loaded error free.

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38379 (Git)
Build type: Release
Branch: main
Hash: c4bb689a851b16d9a22f085e7406db9dd4cc76dc
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
```

Looking at the code the condition to generate that error is found at https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/Tree.cpp#L3839-L3846
So I don't know which of those conditions isn't being met on your system and why, it'll need a (Windows compiling probably) dev who can actually reproduce the error to investigate.

Carcaman, 2024-07-30:
Thank you! and yes you were right. I'll keep on trying on next devs. Thanks for the effort on making a better FreeCad.

Syres916, 2024-07-30:
@wwmayer sorry to trouble you but can you actually replicate this issue and if not, could it be triggered by locale, I've tried setting FreeCAD language to _Espanol_ but it made no difference?

wwmayer, 2024-07-30:
Works for me without problems.

IMO, the issues are caused by the thrown FreeCAD exception. As a result it jumps out of the function stack to the next catch() block that handles this exception. Obviously, this leaves back some objects without a view provider.

Now the actual question is what causes the exception.

Carcaman, 2024-07-31:
I   tested it in version:

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36596 (Git)
Build type: Release
Branch: main
Hash: 380d452adb5da85f47543a316b6b533f84e829c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
 
And it worked perfect

Carcaman, 2024-08-03:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
 
Last Dev Version, new clean FreeCad Directory in User/AppData/Roaming/, I'm having same issue.
[image]

Carcaman, 2024-08-31:
I'm  Still having this issue

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)

Carcaman, 2024-09-05:
Problem Solved! Changed Freecad 0.22 folder to other location, closer to disk root, and now it's working fine! Problem was it was original located in something like;  D: Folder1/Folder2/Folder3/.....Folder8 or so, very far from root. It was my fault.

closed 2024-09-05 by maxwxyz