FreeCAD/FreeCAD #15653: Copying part of the sketch from one sketch to another doesn't preserve a lot of details

nazar-pc, 2024-07-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried a relatively simple thing:
* selected part of the sketch
* copied selection with Ctrl+C
* opened another sketch in the same plane
* pasted copied contents with Ctrl+V

I'm not sure if this is one or multiple issues, but I noticed these unexpected side effects:
* while constraints were preserved (except against external geometry, which happens to correspond to new sketch could have been preserved trivially), following things were not:
    * expression `0.8 in * 1` was replaced with plain computed result
    * `Name` was cleared
* points that were not construction geometry before copying became construction geometry after pasting (which is even more confusing because there is no visual distinction between them in current dark theme)

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * parts_library
  * Curves 0.6.40
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-28:
Could you provide a demo file.
@PaddleStroke FYI

nazar-pc, 2024-07-28:
Here is a trivial demo, no file is needed:

https://github.com/user-attachments/assets/70821a1c-866e-432b-904d-5cb5e3053b13

kaiwas, 2024-07-29:
I think these are related issues.

https://github.com/FreeCAD/FreeCAD/issues/15579

It seems to me that expressions that pass through the "clipboard" are converted to numbers.

PaddleStroke, 2024-08-05:
Thanks for the report. Indeed #15579 is the same as this one. Though this one report more than one bug actually.

I think the current copy command is only copying the raw value of the constraint, hence loosing any formula.

PaddleStroke, 2024-08-05:
The point turning to construction is due to this: https://github.com/FreeCAD/FreeCAD/blob/6fda739a14b57da0c7d84230c17933738493d5bf/src/Mod/Sketcher/App/SketchObject.cpp#L1565

It is apparantly done on purpose for some legacy code support...

PaddleStroke, 2024-08-05:
The rest of the bugs reported are limitations of Abdullah's PythonConverter.

maxwxyz, 2024-08-05:
@nazar-pc could you update the title and post to reflect the last open issue?

nazar-pc, 2024-08-05:
Please feel free to edit as you see fit

PaddleStroke, 2024-08-05:
Ok so I have a PR up that fix 2 of the bugs : construction point and reference/deactivated being lost.
Next will be the reference being lost.

luzpaz, 2024-08-27:
Updated ticket to show tickboxes for each issue described (we cam tick them off as they get fixed).

closed 2025-07-02 by maxwxyz