FreeCAD/FreeCAD #15655: PartDesign create body/migration/undo/create body causes segfault

nazar-pc, 2024-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I try to create new body in existing document, FreeCAD asks to do a migration that I agree to, then I undo the operation and next body creation causes segfault, here is demo:

https://github.com/user-attachments/assets/0a5c0cb8-e9a9-48ae-88fe-5e65aa16dc67

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * parts_library
  * Curves 0.6.40
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-28:
@FEA-eng didn't you remove the migrate tool recently?

FEA-eng, 2024-07-28:
> @FEA-eng didn't you remove the migrate tool recently?

That’s right, I removed it from GUI.

maxwxyz, 2024-07-28:
So is this tool needed or not? It seems to generate an issue with an older file in the latest weekly.

FEA-eng, 2024-07-28:
> So is this tool needed or not? It seems to generate an issue with an older file in the latest weekly.

From what I know, it’s not needed but it’s also possible that those migration prompts need to be removed as well: https://forum.freecad.org/viewtopic.php?t=89431#p773404

nazar-pc, 2024-07-28:
The file I have was created with 0.22 dev build from beginning on June 2024 as described in above forum post. Never used older FreeCAD versions than 0.22 dev build, which makes all this even more confusing.

nazar-pc, 2024-07-28:
I see, `IoShieldPart` actually contains `PartDesign`'s `Pad` rather than extrude.

No idea how that happened though, it is a `Part`, but it looks like a `Body` and it contains `Origin` and `Pad` like normal `Body` would, I suspect this is why it asks me about migration all the time and gets confused when attempting to do the actual migration.

nazar-pc, 2024-07-30:
https://github.com/FreeCAD/FreeCAD/issues/14799 is likely related to this issue

luzpaz, 2024-08-17:
Any progress fixing this crash before 1.0 ?

maxwxyz, 2024-08-17:
@wwmayer could you help out here about the PD migrate tool which was removed?

maxwxyz, 2024-10-07:
Added the blocker label for discussion of this issue

maxwxyz, 2024-10-17:
@Jookia as you've created the linked issue, can you also confirm this? With which versions were your files created?

Jookia, 2024-10-17:
Hi there,

I don't remember and I no longer contribute to FreeCAD.

Jookia.

PaddleStroke, 2024-10-21:
@nazar-pc can you please provide the minimal file that can reproduce this issue?

nazar-pc, 2024-10-21:
I do not have one I'm afraid, but one of the files from the issues I reported around the same time should be sufficient for reproduction. I can try to find one later, a bit busy right now.

nazar-pc, 2024-10-21:
Try this one: https://github.com/FreeCAD/FreeCAD/issues/15738

If not I'll extract the version as of the day of this report from backup.

PaddleStroke, 2024-10-21:
I tried this file and get no prompt to migrate. That file does not have compounds as in your video.

maxwxyz, 2024-10-21:
@nazar-pc can you reproduce it with the file in the latest weekly?

nazar-pc, 2024-10-21:
Restored older version from backup, errors and segaults exactly the same.
Files (open `Core.FCStd`): [fctmp.zip](https://github.com/user-attachments/files/17464642/fctmp.zip)

```
OS: Ubuntu 24.04.1 LTS (Custom/custom/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git) AppImage
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

PaddleStroke, 2024-10-23:
I can confirm this crash.
But how come do you have a part design feature inside a App::Part ?
[image]

The solution I see is to remove the workflow testing alltogether. As @FEA-eng mentioned before it is old and unlikely to be still useful. And the migration script is triggering in annoying places apparantly.
@chennes thoughts on that? If it's ok then I'll make a PR to remove those.

PaddleStroke, 2024-10-23:
Here is the simplified file that can reproduce the crash. @nazar-pc ideally when you post an issue please make sure to attach a file, and if possible as simplified as possible while still reproducing the crash. It helps debugging faster.

[Core.zip](https://github.com/user-attachments/files/17487677/Core.zip)

nazar-pc, 2024-10-23:
> But how come do you have a part design feature inside a App::Part ?

I have no idea how I ended up with that, I can just say that I didn't use any versions of FreeCAD older than weekly builds from June, so it was (is?) possible to do in the app.

I was very new to it and probably clicked something that more experienced users wouldn't, but no way I can recall it in this case unfortunately.

closed 2024-10-25 by chennes