FreeCAD/FreeCAD #15659: A simple loft results in an invalid solid

Panda631, 2024-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I did a simple loft, based on two (identical) sketches.

[image]

Then, run check geometry with preselected settings.
The result is an invalid solid...

[image]

...and a couple of error messages.

[image]

Btw:
**Opening in v021**,  reports, that the sketches are not attached to anything.

[image]

The model:

[Loft-v22.zip](https://github.com/user-attachments/files/16402700/Loft-v22.zip)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.21
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.1.23
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-28:
Can confirm with the same version.

knipknap, 2024-07-28:
Cannot reproduce in the following version:

```
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git) AppImage
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * DynamicData 2.62.0
  * btl 0.9.9
  * WebTools
  * toSketch
  * fasteners 0.4.62
  * lattice2 1.0.0
  * freecad.gears 1.2.0
  * sheetmetal 0.4.23
  * CycloidGearBox
  * parts_library
  * Assembly4 0.12.3
```

The verification shows no errors.

maxwxyz, 2024-07-28:
@wwmayer you've made the last changes to PD Loft, any idea about this?

wwmayer, 2024-07-28:
> you've made the last changes to PD Loft

No, I didn't. This is the latest change 6a5207f7a889461 and the change I made was to fix a compiler warning.

Jameson2021, 2024-07-28:
I opened loftv22 in 0.21 & saw the same blank attachment feild - & again in 38334 & the attachments were also fine. Like my problem below, it works as it should in v22 if you go back to 36904 also

Similar issue (sketch attments/lofting/versions ) Not trying to Hijack, but was about to post a query of potential bug & saw this {same day)

I've had a full day 7 hours trying to work out a similar additive loft issue (windows) & created a new test file to see what it was getting sensitive about
on 36904 (&v21.0 & v21.2) it (my full project) was working & it (my test file & now panda's v22loft) still does, 
38334 it does not -FWIW neither full project nor small test, nor pandas
I tried 37730 because I had one ready to fire - also no joy

19:40:28  **AdditiveLoft001: Wires are not coplanar.** - Body000 is Xy plane and fine, YZ or XZ planes give the error.
===
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git)
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * CfdOF 1.27.4
  * freecad.gears 1.0.0
  * MakerWorkbench 1.0.1
  * Render 2024.7.4
===

[000 Test.zip](https://github.com/user-attachments/files/16404889/000.Test.zip)

Here's my test file 
Just refresh to kick the pending actions and loft in body001 breaks while the XY plane is fine

If you're looking what broke & when, here's panda's working Ok in 36904

[image]

chennes, 2024-08-14:
I believe this is fixed by the recent TNP work by @bgbsww -- I cannot reproduce in a clean build from today's source. @Panda631  and @maxwxyz can you re-check?

Jameson2021, 2024-08-14:
Sure, will do – thanks for the notification

Regards

James
Director, Design & Materials
Advanced Trimarans Limited
Website: https://www.AdvancedTrimarans.com<https://www.advancedtrimarans.com/>
***@***.***

128 City Road, London, United Kingdom, EC1V 2NX
Company number 14944340<https://find-and-update.company-information.service.gov.uk/company/14944340>

maxwxyz, 2024-08-14:
I have no issue anymore with 
Loft works as expected, sketches have no issues and part check is valid.
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

closed 2024-08-14 by maxwxyz

Panda631, 2024-08-14:
I can confirm as well.
Works now perfect.